betaflight/betaflight #3394: Weird pitch issue

dfarmer0033, 2017-06-30:
[image]

As you can see the gyro shows a spike right after throttle is cut. Any ideas on what could be causing this?

Cheredanine, 2017-07-02:
Could be wrong, but this looks like normal behaviour and what antigravity mode is designed to control
Have you tried it?

dfarmer0033, 2017-07-02:
Yes I have tried anti gravity. Set the gain to 10 and made sure it was turned on as a feature. No change. 

The part that looked weird to me was the gyro movement. The punch was from hovering to full throttle. The gyro showed the movement that happened but why would the quad pitch so hard forward. Possible flight controller issue?

dfarmer0033, 2017-07-02:
I believe the issue has to do with the version of Betaflight I was using. I was running version 3.1.7 on a Lumenier Lux V2 when the issue started  I am now running version 3.1.5 and the issues are completely gone.