betaflight/betaflight #3397: new tiny VTX with onboard serial OSD - best way to implement betaflight support?

fishpepper, 2017-06-30:
Hi!

I am working on a next generation OSD right now. It is not based on the max7456 chips and has 
some more features with many already on my todo list. 
A first proof of concept video can be found here:
https://www.youtube.com/watch?v=USWiuCVAzIQ
The first design will be a 16x16 vtx with integrated OSD. Both, the vtx and the osd will be fully configurable via a serialport. A first (working) draft of the serial osd protocol (max7456 inspired for now) can be found here:
https://github.com/fishpepper/betaflight/blob/tinyOSD_stable/src/main/drivers/tinyosd.c

Before I start to work on the vtx configuration protocol I wanted to know what the best way would be.
Right now I can think of two options:
A) separate vtx and tinyosd serial protocol, both can be assigned to the same serial port (sharing)
B) vtx and tinyosd in one driver

In any case the max7456 osd is quite closely coupled to the board hardware defines right now 
(e.g. using ifdef definitions to switch on osd support etc).
I would like to have this selectable via runtime configuration, so this would require some work as well.

So I am asking the betaflight core devs: What would be your favorite A,B, or a C that I have not yet thought of?
Looking forward your ideas :)

DieHertz, 2017-06-30:
I think they should be separate.
With the current quest for unified targets we'll make MAX7456 configurable at runtime rather than compile time at some point, so that shouldn't be an issue.

Keep up the good work!

martinbudden, 2017-07-01:
I agree they should be separate.

Note also that the current OSD code is intended to be independent of the underlying OSD hardware. The separation is not entirely complete yet, but that is the intention. The idea of this was to allow other OSD hardware in future, so if some work is required to allow tinyOSD configuration, then that is sort of part of the overall grand plan.

The OSD code in `io/osd.c` doesn't talk to the MAX7456 directly, it works via a `displayPort` which abstracts the screen device (analogous to a serial port which abstracts a serial device). A `displayPort` can also be regarded as analogous to a Drawing Context in graphics programming.

You say tinyOSD supports graphics. That will require adding graphics functions to the displayPort vtable, perhaps functions like `drawLine` and `drawRect` - currently the displayPort only supports character devices, but the intention always was that it could be extended later to support raster devices. So feel free to add graphics functions to the displayPort vtable.

ledvinap, 2017-07-01:
@fishpepper : 
You should split your protocol into layers - for example `tinyOSDDrawScreen` should simply receive buffer to fill.
Another layer is responsible for transmitting the buffer - it may prepend header and append checksum and sent it over serial line, or it can be passed to MSP with its own framing.

This will force the code to work packet-wise - lower layers handle flow control - which does complicate things a bit. But it blends very well with CF 'multitasking' system.

Another consideration is drawing primitives - there will be a lot of small commands necessary to update OSD screen (this will get worse if graphical display is implemented).
Packing multiple `draw` commands into single protocol-level packet may considerably improve performance and allow easier further extension.

You should also decide how to implement extended functionality:
- OSD may be implemented as generic display device that supports drawing primitives and all display logic implemented on flight controller. This will make it easy for CF developers to implement new OSD features that will be supported with all compatible OSD devices.
- OSD may be 'intelligent', receiving FC state (attitude/voltages/flight modes/...) and fully handling display layout, graphics etc. Multiple OSD will have it's own implementation, it will be a bit harder to coordinate development.
- Mixed approach - I'm afraid  that that this can easily mix disadvantages of both variants above. There will be tens of 'widgets' (command 0x07), with slightly incompatible or incomplete implementations. It will be hard to make OSD configurable, but working on multiple display devices.

BTW: Can you make PR for your branch? It will be possible to comment actual code ..

fishpepper, 2017-07-01:
Thanks for the comments so far! Good to hear that you are interested in getting the max7456 stuff runtime configurable in the future.

@ledvinap:
Right now the whole drawscreen stuff etc is borrowed from the 3.1.7 max7456 implementation.
I think the intelligent OSD that does the actual drawing itself would be the best (performance wise).
But harder to configure and indeed, it looks quite hard to coordinate the integration in bf at my current stage of development.
For now I will go the easier way by staying "backwards" compatible to the max7456 for the normal OSD stuff with some "intelligent" additions like the stick position rendering. So basically the mixed approach :-\
E.g. My OSD would basically allow free x/y pixel positioning of text. But that would require a lot of changes of the configurator as well. As long as I am busy with the OSD hw and fw I simply can not start to work on that front as well *g* 

Concerning graphic code: Currently the full graphic stuff refers to the (custom) Logo that is being rotated, the pilot logo, and the custom stick gui. The logos can be uploaded to flash by the user. The stick "widgets" are also rendered on the OSD. Custom line drawing etc will be added later (I think of online FFT overlay).
Not sure how much performance I will need on the OSD for real drawing functions on the FC.
Most likely the drawing will be implemented on the OSD and will fed by a FFT binary buffer dataset.
The char "drawing" stuff from the max7456 code is also quite intelligent already, it only sends the changed chars. For my code I added some stuff to send "connected" changes in one transfer.

@martinbudden: Concerning max7456 independence: In addition to those places with #ifdef MAX7456/OSD 
there are e.g. also hard coded char buffer sizes and positions. 
As my OSD uses 35 chars instead of 30 per line the default positions where the different things pop up are wrong (crosshatch). So I will have to make those configurable or osd hardware dependent.

Splitting OSD/VTX: That what was I thought to be the best as well.
I have no (good) idea how to handle the dual use access of the serial port in a clean way.
The vtx commands will be in the same frame format as used by the OSD code.
I will have to add a mutex somewhere to orchestrate the dual access to the serial port in order to make sure that the protocol is not messed up :-\ Are there any examples where the serial TX line is shared among two "drivers"?
What about abstraction layer / driver that provides osd and vtx setting calls and the OSD and vtx drivers both use this "virtual" port driver instead of doing raw serial port access?

How should I start the (proper) integration? 
I think the best start would be to have the bf code #ifdef OSD and #ifdef MAX7456 free and to have
the osd stuff on/off configurable via software. Is anybody already working on that?

PR: right now it is based on 3.1.7.  It's hard to stay on the ball when I see how fast betaflight evolves ;)
So it makes no sense to submit this, or does it? Anyway it was not meant to be merged into bf in this way as it is more or less a dirty hack ;)

jflyper, 2017-07-02:
> Splitting OSD/VTX: That what was I thought to be the best as well.
I have no (good) idea how to handle the dual use access of the serial port in a clean way.
The vtx commands will be in the same frame format as used by the OSD code.

A generic "Peripheral device bus protocol", over serial or I2C 😉

martinbudden, 2017-07-02:
WRT OSD item positioning etc - yes these are currently hardcoded. However `displayPort` has `rows` and `cols` members (which should be renamed `rowCount` and `colCount`) - they can be used for positioning thus removing this dependency on MAX7456.

fishpepper, 2017-07-02:
Ok, I will create a separate pull request that uses those variables to calculate the current hardcoded default positions as a start.