betaflight/betaflight #3406: SmartAudio unsupported on F7 targets

hydra, 2017-07-01:
SERIAL_BIDIR_PP support needed.

See - smartaudio serial port flags.

https://github.com/betaflight/betaflight/blob/master/src/main/io/vtx_smartaudio.c#L684

F3 code (with SERIAL_BIDIR_PP)
https://github.com/betaflight/betaflight/blob/master/src/main/drivers/serial_uart_stm32f30x.c#L181-L185

F7 code (SERIAL_BIDIR_PP missing)
https://github.com/betaflight/betaflight/blob/master/src/main/drivers/serial_uart_stm32f7xx.c#L343-L358

blckmn, 2017-07-01:
https://github.com/betaflight/betaflight/pull/3407

Have yet to test it.

jflyper, 2017-07-02:
@hydra It has a reminder already 😉  #3090

hydra, 2017-07-02:
@jflyper ok, well spotted.  I'll close #3090 as this has more detail now.

jflyper, 2017-07-03:
Ack.

digaus, 2017-07-03:
Did this only apply to SmartAudio or also to Tramp Telemetry? Because I had the Tramp working already. Now I have a TBS Unify on the way :/

jflyper, 2017-07-03:
@digaus It is only required for SmartAudio. Tramp uses regular BIDIR.

digaus, 2017-07-08:
Mhh I have 1c1cebc build on my F7 and SmartAudio is working without any problems (just got the Unify Pro Hv) ??

jflyper, 2017-07-08:
It has been fixed by https://github.com/betaflight/betaflight/pull/3407

I don't know the temporal relationship with 1c1cebc, though.

closed 2017-08-03 by jflyper