betaflight/betaflight #3411: VBUS_SENSING_PIN is not used

jflyper, 2017-07-02:
`VBUS_SENSING_PIN`, defined in many F4 (and a couple of F3) targets, and initialized in `vcpf4/usb_vcp.c`, but not used at all.

https://github.com/betaflight/betaflight/blob/master/src/main/vcpf4/usb_bsp.c#L83-L85
```
 83 #ifdef VBUS_SENSING_ENABLED
 84     IOConfigGPIO(IOGetByTag(IO_TAG(VBUS_SENSING_PIN)), IOCFG_IN_FLOATING);
 85 #endif
```

It could be used in `beeper` to suppress beeping under `-ON_USB` when lipo is simultaneously connected. (The current code looks at cell count and voltage.)

Our options would

1. Remove everything related with `VBUS_SENSING_ENABLED` and `VBUS_SENSING_PIN`.

2. Add a code to use `VBUS_SENSING_PIN` in beeper.c. (We might have to create `drivers/vbus.c` if we don't want to modify `vcp4/usb_bsp.c`.

3. Take it as a provisional feature and leave it as is, but remove `VBUS_SENSING_ENABLED` and `VBUS_SENSING_PIN` from target.h.

mikeller, 2017-07-02:
Do we know if all of the boards that define `VBUS_SENSING_PIN` are actually wired so that it is working? I suspect rather not.

jflyper, 2017-07-02:
Personally, I would like to take this opportunity to check out how many of them are really wired 😜 

In reality, we can initialize it as input pull down so that sensing for High defaults to Low if it is not wired (open) and the behavior of the beeper doesn't change.

mikeller, 2017-07-02:
That, or if we find the current hardware does not support it (and nothing seems to be making use of it anyway), we should remove it.
Pins (and their layout constraints) are valuable, so I don't think there will be much uptake for something that does not have any bearing on flight behaviour, at the cost of more layout constraints.

brianlbalogh, 2017-07-02:
USB_DETECT_PIN is only defined for a couple targets, but it's actually implemented in https://github.com/betaflight/betaflight/blob/master/src/main/drivers/usb_io.c  It appears to only be used for reducing transponder current draw at the moment.

jflyper, 2017-07-02:
@brianlbalogh Good find. And `USB_DETECT_PIN` is different from `VBUS_SENSING_PIN`...

OMNIBUSF4 has them both 😮

brianlbalogh, 2017-07-02:
Didn't notice that one, interesting.

jflyper, 2017-07-04:
Oh ... this is bad ...

https://github.com/betaflight/betaflight/blob/master/src/main/drivers/usb_io.c#L65-L76
```
65 void usbGenerateDisconnectPulse(void)
 66 {
 67     /* Pull down PA12 to create USB disconnect pulse */
 68     IO_t usbPin = IOGetByTag(IO_TAG(PA12));
 69     IOConfigGPIO(usbPin, IOCFG_OUT_OD);
 70 
 71     IOLo(usbPin);
 72 
 73     delay(200);
 74 
 75     IOHi(usbPin);
 76 }
```

PA12 is magic... it's meant to be USB_DP ...

mikeller, 2017-07-05:
I think this comes straight out of the STM reference code...