betaflight/betaflight #3416: Omnibus F7 ICM20608G  - Roll issue

sdotfpv, 2017-07-02:
While trying to experiment with 32k 32k fc locks up at %100 which i know is still normal and devs are working hard on that. The best i can run on the Omnibus F7 atm is 16k gyro and 8k pid loop i get about 24-25% cpu usage i calibrate the gyro a few times be4 going to the feild after about 3 packs in mid flight if i level out on the roll axis the kwad rolls to the left so  i constantly have to manually level it back after unplugging and plunging bat it stops for a a min or so but comes back throughout the next battery. Absolutely 0 issies while running the MPU6000 gyro at 8k 8k  - also with the bf osd i can clearly see that my sticks are centered and endpoints in the taranis are also corrected

nathantsoi, 2017-07-02:
can you post the blackbox?

sdotfpv, 2017-07-02:
also part of the problem which im not sure if its related but my sd card now is showing write protected cant format or add any new data. not sure if the last data saved is included but im waiting on a new sd card and will get it one way or another

hydra, 2017-07-02:
@sdotfpv replace the sd card, sd card's do that when they detect unrecoverable errors so you don't destroy any data on the card.

sdotfpv, 2017-07-05:
yea ive got the new sd card coming today but ive also updated the firmware since then so ill post the next avail BB log

sdotfpv, 2017-07-07:
So i just updated BF to the latetest 1431  set gyro and selected 16k 8k and now i get 42% cpu usage instead of the 28 i was getting with a previous build

sdotfpv, 2017-07-08:
@hydra so replaced the sd card this is now the 3rd sd card that is showing write protected after formating no problem and placing into the sd card port. no idea now how to remove the write protection on this brand new sd card

sdotfpv, 2017-07-16:
Here is the blackbox data also included the vid to go along

sdotfpv, 2017-07-16:
[LOG00002.zip](https://github.com/betaflight/betaflight/files/1150836/LOG00002.zip)
Log

sdotfpv, 2017-07-16:
cant add the video file to large ill try to make it smaller

sdotfpv, 2017-07-16:
@nathantsoi 
[Blackbox1.mp4.zip](https://github.com/betaflight/betaflight/files/1150839/Blackbox1.mp4.zip)
video file
[LOG00002.zip](https://github.com/betaflight/betaflight/files/1150840/LOG00002.zip)
blackbox log

closed 2018-04-21 by DieHertz