betaflight/betaflight #3431: BF3.2.0 - reboot to bootloader is broken

TheIsotopes, 2017-07-04:
since [jenkins build 1420](http://andwho.sytes.net:8080/job/BorisB_BetaFlight/1420/) is the reboot to bootloader on my revolt f4 broken ... cli `bl` and the button `flash firmware` only reboots but not in bootloader ... imho the suspicious seems to be #3355 ... on my f3 boards i could not determine this problem

Faduf, 2017-07-04:
#3355 only modified the startup and System Init files for F40x boards. There should be no modifications in F3 Firmwares.

TheIsotopes, 2017-07-04:
tested your fix #3432 ... the issue is now solved

TheIsotopes, 2017-07-04:
ok, it's merged ... imho i can close now

TheIsotopes, 2017-07-05:
after revert #3432 from @borisbstyle and testing latest master the issue is now back

mikeller, 2017-07-05:
@TheIsotopes: If you are using the boot pins / button on your board, does it go into DFU mode? If not, then your board is borked, and no firmware change will be able to fix that.

TheIsotopes, 2017-07-05:
@mikeller are you read my op?
the boot pins are working ... its an firmware issue

mikeller, 2017-07-05:
Yes I did - and misunderstood. I thought '...and the button...' meant the hardware button, not the button in the configurator. My bad, apologies.

jflyper, 2017-07-05:
Stand-by, fix is coming.

closed 2017-07-05 by borisbstyle

TheIsotopes, 2017-07-05:
@jflyper tested your revolt.hex
finally all is working now

jflyper, 2017-07-05:
Good to hear 😄

commit 4b01b9d4 (2017-07-05): Merge pull request #3444 from jflyper/bfdev-fix-issue-3431

Fix #3431