betaflight/betaflight #3433: Death Rolls, Airmode, 6x4.5 Props, BF 3.1.7

MWolschlager, 2017-07-04:
Hello! I've been having some issues with "death rolls" in airmode with BF 3.1.7. I put that in quotes since it's not a true death roll, I'm usually able to "catch" the quad but only after it's done 2-3 rolls. 

Configuration below: 
- Frame: QAV-R 
- FC: RMRC Dodo F3
- 4kHz/2kHz, DSHOT600, Airmode always on (also tried on a switch same result), Motor Idle tested between 2-8% with no different results 
- X4r-SB
- Motors: Brother Tornado T2 (switched to Cobra 2100kV race to see if motors were a problem)
- ESC: Spedix IS 30A (unfortunately didn't have a different brand of ESC but I replaced what was on them with brand new ones of the same type. Also I have a Raceflight build that uses these on Multishot just fine)

Originally started off with the Tornado T2s the death rolls only came out when I raised my rates to: RC Rate 1.00, Super Rate 0.78, RC Expo 0.00 on stock PIDs. I lowered PIDs to try and solve the problem, which helped, but death rolls would still happen every so often. I had to lower my PIDs so far down (down to PID: Roll: 6, 5, 7 Pitch: 10, 5, 9 Yaw: 15, 10)  that the craft wasn't stable before they stopped entirely. 

Looking through forums I found that it may be due to a bad flash: I re-flashed FC completely set everything to stock and slowly bumped up rates. Once I hit super expo to 0.78 death rolls started happening again. 

I then tried the exact same setup but with 5x4x3 props: I can reliably move Super Rates up to 0.85 and everything flies smoothly on stock PIDs (blackbox log included below). Throw 6x4.5s back on and death roll central.

I also found online that it may be hardware related:  I changed my ESCs to new ones (same brand and type though); no change. I tried playing with Demag and Motor timing no difference. I then switched to the Cobra motors listed above; even worse, death rolls happen every time now. 

I think the BB Log labeled "New Motors" really shows what's happening. Look at 9/9 of that log and watch the second roll with PIDs, rcCommand, and Gyro showing. 

1. - I do a roll (that death rolls), I end up upside down at 17.3s to 17.7s (BB Log craft display shows this as a single roll landing right side up but I can confirm that it was more than that and I ended up upside down)
2. - I then end up slightly rolling again to correct myself to right side up 17.7s to 18.0s (BB Log craft display shows this as flipping myself upside down), 
3. - at 18.0 is when I start re-applying throttle to keep myself in the air (if I was truly upside down like BB Log shows I would have accelerated into the ground for two seconds from 18.0s to 20.0s and since I was only flying LOS I would have slammed into the ground. I didn't because I wasn't upside down.)
4. - At 20.0s the BB Log Craft display just randomly rights itself. No RC command no gyro movement just right side up again. 

The only thing I have left to try is replacing the FC or trying different brands of ESCs. Seeing as I've used this same FC in the past with older BF builds with Airmode I don't see why it would be a problem. As far as the ESC's go, they're brand new ESCs that have been tried with two different motors and they work just fine on my other build which is also running the same motors. 

Is there anything you might be able to suggest to try and solve this problem? 

Thanks!
[BF 3.1.7 Black Box Logs.zip](https://github.com/betaflight/betaflight/files/1122772/BF.3.1.7.Black.Box.Logs.zip)

chime13, 2017-07-14:
Flash with 3.1.6 or 3.1.5. 
3.1.7 is known to cause these issues on many builds.

MWolschlager, 2017-07-14:
Seems like the flight controller was on its way out. I tried to flashback to 3.1.6 and it "bricked" itself. Wouldn't install firmware wouldn't let me back into betaflight setup. Tried jumping boot pins no success. Got a Betaflight F3 and all is right with the world.

closed 2017-07-14 by MWolschlager