betaflight/betaflight #3437: BF3.2.0 - reboot to bootloader is broken #3432, OMNIBUSF4 PRO

AirBreak69, 2017-07-04:
I flashed the latest Jenkins build #1425 containing patch #3432 into my DYS F4 PRO (OMNIBUSF4).
Now it is completly dead! Jenkins build #1419 was the last working build i tried.
No connect is possible, no DFU via boot switch.
I have to mention: The board was slightly damaged some weeks ago by overcurrent in USB-only operation. A design flaw. The board should only be connected to USB when a battery is connected.

With the builds before I had to invoke "bl" in CLI 5-10 times to get into DFU. The boot switch is useless since that overcurrent situation. The board locks up without DFU if started with boot switch.
Help wanted!

mikeller, 2017-07-04:
The firmware is not involved in booting into DFU mode through the boot pins / button. So if this does not work any more, that means your board is most likely toast.

jflyper, 2017-07-05:
@AirBreak69 
The original OMNIBUS F4 AIO with latest master does not have the issue you described.

Please consult other sources for support.

TheIsotopes, 2017-07-05:
are you overclocked your omnibus f4?

jflyper, 2017-07-05:
@here The FC wedges on boot when `cpu_overclock = ON`.
Confirmed with the original OMNIBUS F4 AIO with #3432 applied.

Related: #3442

jflyper, 2017-07-05:
@AirBreak69 Can you change the title to "Overclocked F4 wedges on boot" or something similar?

borisbstyle, 2017-07-05:
Reflash with the lastest master and it will be fine

closed 2017-07-05 by borisbstyle

TheIsotopes, 2017-07-05:
@borisbstyle are you sure 
imho reverting the patch brings back the issues with no running bootlader

jflyper, 2017-07-05:
Yes. We are back in #3431.

- `bl` does not invoke DFU mode.
- Hardware BOOT0 jumper/button is okay.

TheIsotopes, 2017-07-05:
i have reopen #3431

AirBreak69, 2017-07-05:
@TheIsotopes no i didn't activate overclocking. I simply flashed build 1425 with patch #3432 over 1419.
@borisbstyle because my board cannot be accessed anymore, (the board doesn't boot) and the boot switch does not bring it to DFU, i cannot flash it via USB.
If there isn't another way to flash a F4 via USART by an FTDI adapter, i have to accept that the board has to be changed.
A very bad combination of negative events and a bad board design where some components can be grilled if USB 5V can go backwards into the unsupplied BEC converter...

jflyper, 2017-07-05:
@AirBreak69 The boot switch/button doesn't work??? Do you have any active component connected to UARTs when you do the DFU boot? If so, they may fool the boot loader and capture the false focus. Powering them down or disconnecting them may help.

AirBreak69, 2017-07-05:
@jflyper Giant!!! What a brilliant hint!
The GPS receiver connected to UART6 disturbed the DFU boot.
I compiled the latest PR and was able to flash it instantly.
That saved my day, some money spent in China and weeks of waiting.
Thanks a lot!

jflyper, 2017-07-06:
@AirBreak69 That is good to hear!!! 😄

AndersHoglund, 2017-07-06:
I have similar problem with Spektrum AR4648. 
When in Autobind mode, fast blinking LED, it seems to flood the boot loader making the bl non-responsive. Seen with SPRF3, that is not even a VCP USB FC, UART connected USB. I have to dis-connect satellite for boot loader to work.

cr3pitus, 2017-12-17:
I flashed bf to my revolt and went to set cpu_overclock = on and had this same issue.  What worked for me was to disable to uart the receiver is on prior to turning on the overclock.

mikeller, 2017-12-18:
@cr3pitus: This is expected behaviour. The boot loader listens for input on multiple UARTs and VCP, and responds to the first port that gets input. If your RX is sending data while the flight controller is booting into the boot loader, this will be triggered, thus causing the boot loader to stop listening on VCP. That's why there is a note on the firmware flasher page:

> Note: If you have problems flashing try disconnecting all cables from your FC first, try rebooting, upgrade chrome, upgrade drivers.