betaflight/betaflight #3442: BF3.2.0 - revolt f4 no responds after cpu overclock

TheIsotopes, 2017-07-05:
i wanted to test the cpu overclock an my revolt f4 ... used 32/32 and `set cpu_overclock = ON` but after save/reset the fc makes no responds ... no initial beeps and no usb ... resurrection is only possible after flashing bf3.1.7 with shorted boot pads

@Faduf pls can you see what's wrong

jflyper, 2017-07-05:
@TheIsotopes Can you close this and continue on to #3437?

It's the same issue, and I think it's better to consolidate related threads into one rather than spreading comments among multiple threads.

TheIsotopes, 2017-07-05:
ok ... closed now ... continue on #3437

closed 2017-07-05 by TheIsotopes

borisbstyle, 2017-07-05:
Reflash to latest master and the baord will be fine

NIHERASE, 2018-11-17:
I did a `set cpu_overclock = ON` on my kakute f4 v2 and I think it's completely bricked now. Can't reflash with boot button. `COM` device does not appear.