betaflight/betaflight #3448: X_RACERSPI not working after commit 7a52f91

KrickeH, 2017-07-05:
issue is verified by jflyper and he have done a bunch of test firmwares to isolate bug, but we can not get it to work without using fake gyro and acc.

jflyper: please feel free to add or edit as you please.

jflyper, 2017-07-05:
I'm suspecting somewhere along SPI2~flashchip~flashfs line is hard faulting.

DieHertz, 2017-07-05:
@jflyper remember when you had issues with `uartDevmap[]` and `uartDevice[]`?
They have finally shot us in the foot. `drivers/serial_uart_pinconfig.c` doesn't like to be size-optimized, we're probably violating one-definition rule or something.

Removing `drivers/serial_uart_pinconfig.c` works around this issue but I feel we should find an explanation what have we violated to get such behaviour.

DieHertz, 2017-07-05:
Both `uartDevmap[]` and `uartDevice[]` do not look right on F3 where size-optimized files are treated separately.
[image]

jflyper, 2017-07-06:
@DieHertz Wow. So it is just a coincident that it hit X_RACERSPI?

jflyper, 2017-07-06:
How about isolating the data structures out of size optimized file and make it non-optimzed, as a pragmatic solution? (Keep init code size optimized while run time code speed optimized without hitting this bug.)

@martinbudden @ledvinap @blckmn Thoughts?

martinbudden, 2017-07-06:
If we are just talking about `drivers/serial_uart_pinconfig.c` then I would be inclined to just turn of size optimisation for that file. It has only one function so it seems hardly worth splitting.

Unfortunately the GCC docs are pretty vague about size optimisation:
> -Os
Optimize for size. -Os enables all -O2 optimizations that do not typically increase code size. It also performs further optimizations designed to reduce code size.

As an experiment, I turned of size optimisation for:
```
drivers/bus_i2c_config.c \
drivers/bus_spi_config.c \
drivers/bus_spi_pinconfig.c \
drivers/serial_escserial.c \
drivers/serial_pinconfig.c \
drivers/serial_uart_init.c \
drivers/serial_uart_pinconfig.c \
```

This cost 544 bytes of ROM for BFF3 build.

So I vote we turn of size optimisation for all the above files.

fishpepper, 2017-07-06:
While looking around how the optimizing breaks this I came around #3449
Should not have to do with this. But who knows, maybe the optimizer is doing soemthing bad here because of that.

martinbudden, 2017-07-06:
Well, it could be that without optimisation the pointer was pointing somewhere harmless...

ledvinap, 2017-07-06:
@DieHertz: Can you post your pSerialPinConfig?
It seems that something is overwriting uartDevMap.

ledvinap, 2017-07-06:
Ad optimization:
- setting optimization on file level is equivalent to function-level setting with LTO enabled - compiler pass just produces GIMPLE representation of source code, no optimization is done. And all optimization is done during linking, with different per-function optimization levels (triggering 'unsupported for production level' notice in GCC docs).
- it is unlikely that gcc will generate invalid code. There was one case in past (USB), but that was related to volatiles and register access, the code generated code was 'only' accessing correct memory with incorrect instruction.
- speed optimization is IMO overrated
- splitting modules to allow different optimization level is IMO crazy - modules should be grouped logically, splitting it by optimization will only make maintenance harder
- (function level optimization can be easily controlled from Makefile - just decorate functions as 'benefits from speed optimization',  'almost never called', 'on critical path' and use `-D` from Makefile + macros to set correct optimization)

jflyper, 2017-07-06:
> It seems that something is overwriting uartDevMap.

I had this impression, too.

ledvinap, 2017-07-06:
@DieHertz : Can you try write breakpoint at uartDevmap ?

DieHertz, 2017-07-06:
Sure I will when I get back home, if it's still relevant.

jflyper, 2017-07-06:
0x20002c22 probably is inside spiPinConfig_Copy or near...

DieHertz, 2017-07-07:
At least some of stomped uartDevmap's members are pointing to spi-related objects.

jflyper, 2017-07-07:
I think I nailed it.

It's in `uartPinConfigure`, where outer most goes around as many as UARTDEV_COUNT_MAX (5 for F3), while `uartHardware` only defines what has been declared as used (3 for X_RACERSPI); last two loops work on something that does not define UART hardware. :sob:

mikeller, 2017-07-07:
Good find, @jflyper.

fishpepper, 2017-07-07:
good find!
`const uartHardware_t uartHardware[UARTDEV_COUNT]`

So it is indeed related to #3449. When uartHardware[3+4] point to random memory the if() conditions might trigger additional times.
This will cause uartdev ptr to exceed bounds (uartDevice[UARTDEV_COUNT]) and will also write (!) to the memory region outside of the uartDevice array.

DieHertz, 2017-07-07:
You have robbed me off a nice debugging pastime for this evening :-) 😞

jflyper, 2017-07-07:
#3453 

@fishpepper Yes and no 😉 It was related, but outside the `uartPinConfigure`.

jflyper, 2017-07-07:
Okay, @KrickeH confirmed #3453 working.

commit b8b1c890 (2017-07-09): Merge pull request #3462 from jflyper/bfdev-fix-bad-uart-pin-initialization-2

Fix #3448 Bad UART pin configuration (Simpler Version)
closed 2017-07-09 by blckmn