betaflight/betaflight #3449: drivers/serial_uart_pinconfig.c - access out of bounds for uartDevice array?

fishpepper, 2017-07-06:
I just found it while looking around for the reason why the optimization causes issue #3448

I was wondering if this code in drivers/serial_uart_pinconfig.c might do something bad for UARTDEV_COUNT < UARTDEV_COUNT_MAX:
```
uartDevice_t uartDevice[UARTDEV_COUNT];      // Only those configured in target.h
uartDevice_t *uartDevmap[UARTDEV_COUNT_MAX]; // Full array

void uartPinConfigure(const serialPinConfig_t *pSerialPinConfig)
{
    uartDevice_t *uartdev = uartDevice;

    for (size_t hindex = 0; hindex < UARTDEV_COUNT_MAX; hindex++) {

        const uartHardware_t *hardware = &uartHardware[hindex];
        UARTDevice device = hardware->device;

        for (int pindex = 0 ; pindex < UARTHARDWARE_MAX_PINS ; pindex++) {
            if (hardware->rxPins[pindex] && (hardware->rxPins[pindex] == pSerialPinConfig->ioTagRx[device]))
                uartdev->rx = pSerialPinConfig->ioTagRx[device];

            if (hardware->txPins[pindex] && (hardware->txPins[pindex] == pSerialPinConfig->ioTagTx[device]))
                uartdev->tx = pSerialPinConfig->ioTagTx[device];
        }

        if (uartdev->rx || uartdev->tx) {
            uartdev->hardware = hardware;
            uartDevmap[device] = uartdev++; // <----- THIS LINE
        }
    }
}
```
Calling this with tx=PC10 for example would match for UART4 tx and UART3 tx (drivers/serial_uart_stm32f30x.c). 
This way uartdev++ could potentially point to some random memory after the array.
uartdev->hardware = hardware might also write to those locations :-\

martinbudden, 2017-07-06:
Good spot. So it appears the problem was nothing to do with optimisation. Although optimisation sometimes brings underlying problems like this to light.

Can you submit a fix?

fishpepper, 2017-07-06:
I can do a fix for the array bounds tonight in the evening. 
In general detecting the uart based on the port+pin seems broken for UART3+UART4.
There is no way to select between instance 3 or 4 based on the input PC10. 
However PC10 is only used on one F3 target (RG_SSD_F3).

But as said I doubt this has to do with the reported issue.
For the XRacer target this loop should only access valid index positions as it does not use UART4.

ledvinap, 2017-07-06:
@fishpepper : Pin is set for given `device`, (`pSerialPinConfig->ioTagRx[device]`).
`pSerialPinConfig->ioTagTx[UARTDEV_1] = PC10`, so it should not match UART4

ledvinap, 2017-07-06:
But array size should be checked ...

fishpepper, 2017-07-06:
Ah right, I did not see that the txPins array is based on the hindex based selection of the uart. 
So this should not cause any issues as long as pSerialPinConfig has no bad data in it (caused e.g. by another bug). 

The array len should be checked anyway. What would be a proper reaction to an out of bounds access?
failureMode(FAILURE_DEVELOPER)?

jflyper, 2017-07-06:
> The array len should be checked anyway. What would be a proper reaction to an out of bounds access?
>failureMode(FAILURE_DEVELOPER)?

We can do it statically.

ledvinap, 2017-07-06:
Actually it shouldn't overflow (unless UARTDEV_COUNT > UARTDEV_COUNT_MAX) ...

jflyper, 2017-07-06:
Yes. But UARTDEV_COUNT depends on number of USE_UARTx in target.h.

As a precaution against the situation, I will be adding the following lines.

```
#if UARTDEV_COUNT > UARTDEV_COUNT_MAX
#error Too many USE_UARTx for the MCU
#endif
```

DieHertz, 2017-07-06:
@martinbudden we knew that it has nothing to do with optimisations, issues rarely do. What rather happens is non-conforming code and undefined behaviour are uncovered by optimizations.

jflyper, 2017-07-07:
#3453

closed 2017-07-17 by jflyper