betaflight/betaflight #3450: 3.2 OSD Issue

ChristAndQuads, 2017-07-06:
When I get into the OSD menu the right stick is not able to move the curser or select anything.  What could be the issue?  Copter seems to fly good.

jflyper, 2017-07-06:
You have to keep left stick dead centered.

For further assistance, please redirect your question to the appropriate thread in RCGroups, as this GitHub is for questions and discussions around the development of the Betaflight firmware.

ChristAndQuads, 2017-07-07:
Sorry if I was in the wrong thread but it worked fine till I loaded the 3.2 firmware. That's why I posted it here. 

I will try again but I believe I am leaving it centered.

voyn238, 2017-07-07:
I thought it was a bug too. Glad I searched a bit and found this, otherwise I'd have had no idea. Already flashed back to 3.1.6 or I'd check with left stick centered.  Why was this added? As some sort of safety thing?

jflyper, 2017-07-07:
> As some sort of safety thing?

Yes. It was added to avoid stick movements while in CMS from being recognized as one of valid stick commands.

voyn238, 2017-07-08:
Like stick-command arm/disarm? Makes sense! I did notice I was still able to change PID profiles via stick commands while stuck in the unresponsive OSD menu. Where's the best place to find out about changes like this? I read the release notes for official BF releases but I'm not sure where to find them for unreleased versions like this.

jflyper, 2017-07-08:
Milestone list may help, but not all of the changes are in there...

voyn238, 2017-07-08:
Alright, thanks!

closed 2017-07-09 by mikeller