betaflight/betaflight #3458: Unable to send MSP SET_RAW_RC commands or use control sticks

BombD123, 2017-07-08:
Hi, coming into this new, so forgive me if there is some history.  I have set RX_MSP, as a feature on my flight controller.  To my understanding, this should mean that I will be giving commands to my flight controller via the multiiwii protocol, and that it will be through UART1.  When this is set, I can see the option to use control sticks in the receiver menu.  But, rather surprisingly, these do nothing.  I've tried this both a NAZE32 board.  I've also then purchased a separate CC3D board and flashed customer firmware that I built with the added USE_MSP_UART flag.  And I have observed the same problem.

Are you aware of any additional steps I can take to enable sending these commands over USB on either of the flight controllers?  Alternatively, is there a way that I can debug what happens when I send receiver commands?  I'm trying to use serial rx with a raspberry pi.  I've separately implemented the SUMD protocol and tried to send it over, but with no success (nor errors).  It feels like I'm sending things over into a black hole and they do nothing.  It sure would be nice if there were error messages or something I could use to guide my intuition here.  Is there anything that could help me?

Thanks,
-Brian

jflyper, 2017-07-08:
@BombD123 Sending MSP_SET_RAW_RC over MSP enabled serial works.
I've done it many times with simple programlets.
I suspect you are sending malformed MSP packets.

DieHertz, 2017-07-08:
Sure you can debug - get yourself an STLink, or debug by blinking with LED(0|1|2)_(ON|OFF) :)

mikeller, 2017-07-09:
@BombD123: The problem with `RX_MSP` is that this protocol has been disabled for F1 based boards (and both NAZE and CC3D are F1 based). In order to get this to work, you will have to build a custom version of the firmware that includes the `RX_MSP` drivers, or else get yourself a more current F3 or better based board (F1 support will be discontinued after 3.2 anyway).

jflyper, 2017-07-09:
Good point ... 😓

BombD123, 2017-07-12:
Got it, okay so in the future it sounds like I should transition to an F3+ model.  For now (in case anyone else runs into the same issue and really needs to use an F1 board), I went into the firmware and disabled all sanity checks for the SUMD protocol / forced it to set values on each channel in order to help me debug.  I was able to observe the values I wanted to be set, but slightly off; it turned out that in my implementation of the SUMD protocol had a mismatch in the endianness of the data being sent.  After reversing, it worked.

I subsequently encountered another intermittent failure where the values in the controller would be picked up correctly for a second or two, and then thrown away.  After some long debugging hours, I found that when I was sending only TX, there was some problem in the serial port picking up the data.  When I hooked up GND and VCC FROM the NAZE flight controller, the problem was resolved and it now works 100% smoothly.  I'll post the code I used to make the controller on my github profile.

Incidentally, CC3D has SUMD disabled.  I figured this was due to size constraints in the firmware, so I tried created a firmware build with SUMD enabled and other receiver protocols disabled so that I could try the same method on it.  It worked to set the allowed receiver protocol, but for some reason I'm unable to entice any of the ports to accept uart input on its main or flexi uart port (and I was unwilling to invest a significant amount of time to debug).  So I gave up on that avenue.

Anyhow, going forward I'll plan to use F3+ on any subsequent projects (hopefully the next one I find will be able to accept uart input more easily, and I can use one MSP port for controlling and telemetry.  Thanks very much for the quick and thorough responses.

closed 2017-07-12 by BombD123