betaflight/betaflight #3465: Omnibus F7 BF3.2 #1435,1436

sdotfpv, 2017-07-08:
When flashing these 2 versions i cannot connect to bf gui the com port doesnt show had to flash back to the last one i had working which was 1409.

sdotfpv, 2017-07-08:
1432 seemed to be the last one the Omnibus f7 was able to recognize the com and connect to the gui

jflyper, 2017-07-09:
@sdotfpv Try 1439. Should okay now.

sdotfpv, 2017-07-09:
@jflyper yea i got it and its working now,  my blackbox though only showing sd card initializing. in bf gui it shows file system starting. when i land and disarm i get a fault on the osd is it just the firmware cant even remember what version i had BB working on maybe 1394

jflyper, 2017-07-09:
Can you elaborate "fault on the osd"?

sdotfpv, 2017-07-09:
So i got a new sd card the ones recommended for blackbox, in the blackbox tab on the gui it shows filesystem initializing. but just hangs there. When i plug a battery in and i arm hover and disarm the bf osd shows the stats like flytime amps used n all that. one of the things is blackbox and i 4get what it shows when its working but since my card is showing initilizing its not recording any bb data.

jflyper, 2017-07-09:
@sdotfpv The "forever initializing" is a known problem for F7.

Have SDCARD logging ever worked before? What was the build number?

sdotfpv, 2017-07-09:
Yea it was working.  i believe last build i had it working was 1391-1394

sdotfpv, 2017-07-09:
ive got 1440 on there now

jflyper, 2017-07-10:
@sdotfpv Okay, I'll do a bisection tonight.

jflyper, 2017-07-10:
Hmm... I can't reproduce the working condition...

sdotfpv, 2017-07-11:
also around the time the blackbox was working had some kind of error locked up my sd card had to get a new one.. so yea dnt even know when the last time it would have been working.. im assuming its working on other targets right?

Alexseii, 2017-09-18:
Help. I can not update the firmware. I do not see the com port. I just insert everything but it does not get fixed

Alexseii, 2017-09-18:
I click boot and can not see the com port 3.2 firmware

jflyper, 2017-09-18:
@Alexseii Sorry, this is not a place for support request. Please redirect your questions and support requests to appropriate places on the net.

Ref:
https://www.rcgroups.com/forums/showthread.php?2884546-Omnibus-f7

closed 2017-09-18 by jflyper