betaflight/betaflight #3470: Omnibusf4 Dys F4 flight controller unable to flash on win10 64bit

NoBadDays, 2017-07-09:
I just got the new flight controller, and while it works, it will not flash on win 10 64bit.  I've tried to install port drivers. I installed the ones betaflight tells you to install. (by the way, the link in betaflight for the cp210x driver and the stm usb vcp driver is for up to win 8.  Says nothing in their respective txt files about win 10 support.  And I think the stm usb is now 3 years old per the text file.

When I hit the flash button after setting everything up i simply get can not open serial port.  This seems to be a none issue if you are using a mac.  Little help here would be much appreciated.

Thanks

NoBadDays, 2017-07-10:
So, not one of you egg heads know how to make betaflight firmware flasher usable on win 10? 
I find that kinda hard to believe.  I know it's a known issue, yet no one here wants to help with it?
And this is why people hate betaflight.

nathantsoi, 2017-07-11:
Hey @NoBadDays. Sorry to hear you're having trouble. Everyone here is helping out, volunteering their valuable time, for free. We'd greatly appreciate it if you would post some logs, dig into the code, maybe do some debugging or even just google a bit more. I'm sure when you find the answer, posting here could be useful to other people.

Do note that every time you post an issue like this, for things that are not real firmware bugs, it takes our time away from making BetaFlight better.

For now, I suggest you post here and ask for help: http://www.dronetrest.com/t/zadig-drivers-for-cleanflight-betaflight-stm32-flight-controllers-dfu-mode/2719

If it does turn out to be a firmware issue, please open an issue against this repo. Also note, that for hardware support, you should contact the manufacturer.

closed 2017-07-11 by nathantsoi