betaflight/betaflight #3475: DYS F4 #1440 No Voltage

riquezjp, 2017-07-09:
Flashed #1440 OmnibusF4 to DYS.F4 board.
Voltage monitoring is not functioning.
Tried both Configurator 3.1.1 & 3.1.2

jflyper, 2017-07-09:
@riquezjp It works on OMNIBUS F4 AIO.

What does CLI `set battery_meter` say?

mikeller, 2017-07-09:
@riquezjp: I am not sure that the pin assignments of the OMNIBUSF4 target are correct for the DYS F4 board. Unfortunately DYS will not provide us with schematics, or even engineering sample boards, so we can do little to find a working target definition for this board.

Try using it with the firmware that is provided by DYS on their website.

riquezjp, 2017-07-09:
battery_meter = ADC

The voltage functions on this board work in bf 3.1.6.

mikeller, 2017-07-09:
What target version of 3.1.6. are you using?

riquezjp, 2017-07-09:
OminibusF4

riquezjp, 2017-07-09:
With this board 3.1.6 works fine, all functions.
3.1.7 has broken support for SBUS
3.2 fixes SBUS support but seems to break Battery - unless its me doing something wrong, but it just looks like every option gives 0v so far.

jflyper, 2017-07-09:
@riquezip
We can't do anything at this point, unless some other information or the real board becomes available.

mikeller, 2017-07-09:
@riquezjp: Is that the 3.1.6 version that DYS distributes with their board? That version is not from us, and we do not have any information on what pins it uses for things like Vbat, since DYS isn't giving us any. Until this changes, you will have to keep using the firmware provided by DYS.

riquezjp, 2017-07-09:
I flashed over the original DYS firmware with 3.2, I found the battery issue. 
So I flashed 3.1.6 as provided in configurator. The battery works.

I cant find DYS firmware on their website, but ill check again ~ but it's useless at this stage as I have blheli32 escs & need to run bf3.2.

mikeller, 2017-07-09:
The definition for Vbat input has not changed for the OMNIBUSF4 between 3.1.6 and 3.2.0, so without information from DYS there is nothing we can do to fix this problem in 3.2.0. Sorry.

riquezjp, 2017-07-09:
You are doing a great job! 
I've dropped an email to DYS to hopefully encourage them to support BF.

jflyper, 2017-07-09:
@riquezjp Since this issue seems to have popped up recently (as there's no conversations on rcg thread), you may want to try bisect (or simple go backward) firmwares at Jenkins to find out at which point the battery meter went dead.

riquezjp, 2017-07-09:
I found the problem!
Voltage monitoring doesnt work when I set Dshot1200. With Dshot600 it works :-o

riquezjp, 2017-07-09:
ahh,well - theres a bug here somewhere.
Now I set Dshot1200 back again & now voltage works.

So - it wasnt working on 1200. I set 600 & it suddenly started working. I set back to 1200 & its still working. odd

jflyper, 2017-07-09:
@mikeller F4 Multi-DMA stream interruption problem?

jflyper, 2017-07-09:
@riquezjp Are you doing full-chip erase when you are going back and forth between different firmware?

riquezjp, 2017-07-09:
full chip erase - yes.

When I first flashed 3.2, from the pre-loaded bf3.1.6, I used configurator 3.1.1
Full chip erase. I downloaded jenkins #1440
I used configurator 3.1.1 to set up everything, sbus, receiver, modes, Dshot1200 in cli, OSD...

At the OSD point I found that in my goggles the osd was full of items & even turning it off (in configurator) it was still loaded up with all the options. At this point i realised about using an updated configurator & got 3.1.2
I cleared up the OSD issues & then noticed 0v

The ADC was set to none so I set ADC & tried a few combination on the two dropdowns ADC/Virtual/ESC to try to get a result, but nothing worked, so i set it back to ADC & started looking online.

I wondered if my board had a problem, so I flashed back to 3.1.6 using the regular target omnibusf4 in configurator. The voltage worked.
I flashed back to 3.2 & found the json backup didnt work :-( so I set everything back up again.
Still no voltage.

Then as I was talking on here I just set Dshot600 using the dropdown, hit save & immediately the voltage showed up in configurator.
I then went to the power tab & the tried to adjust the scale, but the up/down buttons are not active, so i set 109 in the CLI. Now the voltage is spot on.

I set dshot back to 1200 & the voltage continued to work.

I have no idea really, but perhaps me using the old configurator I had managed to set an invalid value for something & it was corrected when i moved to dshot600 standard values set.

pkulka, 2017-07-20:
I have the same issue with no voltage working. I have tried everything you have talk about. Do y'all have any other ideas or am I just missing something

jflyper, 2017-07-20:
Don't restore dump from 3.1.x to 3.2.

Start from scratch.

bizmar, 2017-07-21:
There's something to this. I have a F4 PRO running "DYSF4PRO 3.2.0 Jul 21 2017 / 09:13:41 (9ee4ba4)". I set it up form scratch, copying from screenshots. And my voltage was not working. Then I read the above comments. And switched from DSHOT600 to Multishot. Rebooted and the voltage worked. Switched back to DSHOT600 and it still works. I'll keep you posted if it continues to work. :)

mattkelsey, 2017-08-05:
I'm actually still seeing this in RC2 (specifically the build from August 1st, so it would appear builds past July 30th may not actually fix the issue). I am also using a DYS F4 and DSHOT1200 on the ESCs. I tried the procedures described (switching to DSHOT600, etc) and found none of them to fix the issue. 

The board shipped with 3.1.6 and the voltage worked, I then updated through the configurator to RC2 (which I need for 32 bit escs) and the voltage stopped working. I did a full chip erase flashing RC2 again (with no restore): still no voltage. Flashing back to 3.1.7 fixes the issue.

jflyper, 2017-08-05:
@all
3.2-RC2, tested with engineering samples of "DYS F4 Pro" provided by DYS, has no problem with the voltage meter; it works immediately after re-flashing (enableing and save & rebooting after enabling onboard ADC for voltage, of course), regardless of esc/motor prototocol selection.

I believe the dshot switching procedure described above is not relevant to the battery operation; I suspect it's just invoked a save & reboot required after selecting voltage source.

I think this issue can be closed and relocate to the RCG support thread.

riquezjp, 2017-08-05:
agree with jflyper
I flashed rc2 on a different dys quad & after setup noticed voltage not working immediately. However, save, disconnect, reopen - it was working.

closed 2017-08-18 by jflyper