betaflight/betaflight #3478: betaflight 3.2 osd issue

aviv1993, 2017-07-09:
hey, even i turned off everything in osd on betaflight configurator, i still see crosshair, battery voltage, dsiarmed sign and more ...
i have tried betaflight configurator 3.1.2 and its osd menu isnt working.
how can i fix that ? thanks.
[image]
[image]

mikeller, 2017-07-09:
There's been a lot of changes in the OSD in the last few weeks. Looks like your configurator / firmware are out of synch. Try using latest on both of them, that should work.

aviv1993, 2017-07-09:
is this the latest configurator ? https://github.com/betaflight/betaflight-configurator 
beacuse in this one the osd menu isnt even working.. maybe you can link me to another 3.1.2 version ?
tryed to reflash my 3.2, didnt solved the issue. (i cant flash it on "full chip erase" if it matters)

mikeller, 2017-07-09:
Yes that is the latest version. What version of the firmware did you use? The one in the screenshot seems to be a couple of days old. If full flash erase does not work, try `defaults` in CLI for the same effect, that is what is potentially needed.

aviv1993, 2017-07-09:
can you please link me to the latest 3.2 firmware?
btw, this is what i get on 3.1.2 configurator : 
[image]

DanNixon, 2017-07-09:
What it the git revision of the firmware you are using? (you can get this in the bout menu in the CMS)

It looks like the firmware you have is too old for the latest configurator.

aviv1993, 2017-07-09:
do you mean this ? Betaflight / OMNIBUSF4 3.2.0 Jul  3 2017 / 18:03:09 (5112ffd)

mikeller, 2017-07-09:
Get the latest firmware from http://andwho.sytes.net:8080/job/BorisB_BetaFlight/lastSuccessfulBuild/artifact/

DanNixon, 2017-07-09:
@aviv1993 Yes.

My guess was right, since that version some changes were made to how the OSD is configured and that is likely what the issue is.

aviv1993, 2017-07-09:
Thanks, this fixed the problem. but now my battery voltage is 0.0, is this fixable ?
one more question, if i swapped motor driection in blheli32, do i need to do this aswell : "set yaw_motors_reversed = ON", or swapping the motor is enough.

Thanks.

DanNixon, 2017-07-09:
Do you have a voltage sensor configured in the Battery tab (most likely you need this set on Onboard ADC)?

Yes, if you run the motors reversed you need to configure that in Betaflight too.

aviv1993, 2017-07-09:
no i didnt, and now it works :)
thank you so much for the help guys, your doing great work !

brycedjohnson, 2017-07-12:
Also you can turn on and off features in the OSD itself if you get to the field and accidentally left a bunch of stuff on!

closed 2017-07-13 by mikeller