betaflight/betaflight #3480: Banggood micro FC LED_STRIP and BUZZER

Jahnkeanater, 2017-07-09:
Flashing the board with AirBotF4 firmware fixes the buzzer issue by remapping it to B04 but the problem with the LED pin is that it is connected to the PD2 pin instead of the PA1 pin. If you attempt the remap it. "Resource LED_STRIP 1 D02" the board hangs and requires a reflash to get back into the cli. 
A lot of the new banggood REVO boards have the LED pin attached to PD2 for some reason. Is it possible to remap the LED pin to D02 without the board freezing up?

mikeller, 2017-07-09:
No, unfortunately the PD2 pin does not have a timer and DMA available that is required for LED_STRIP to work, so this pin is probably mostly useless. Another solution is to reassign another pin to LED_STRIP though.

Jahnkeanater, 2017-07-09:
The problem is that this board only has PD2 and PB4 broken out. PB4 is being used for the buzzer.

Is it possible to swap the two pins and use PB4 for the LEDs and PD2 for the buzzer?

Jahnkeanater, 2017-07-09:
LED pin PA1 is way too close to the motor 4 wire PA2 to be able to solder a wire. Is it possible to solder a wire to both PB12 & PB13 and just use one of those? Would there be anything wrong with jumpering the 2 pins? Otherwise, PA14 & PA15 are also easily accessible.

mikeller, 2017-07-09:
PB4 should work, except there normally is a driver FET on the buzzer pin, so the signal will most likely be inverted. Bridging two pins should be fine too, as long as only one of them is configured for output, and the other one is unused.

Jahnkeanater, 2017-07-09:
Well I tried using B12 and then A14 and both result in the board locking up after saving. :/

Jahnkeanater, 2017-07-09:
Alright, I finally got it working by jumpering PC8 and PC9 and using timer3_CH3 on PC8.
Resouce LED_STRIP 1 C08.
:)

mikeller, 2017-07-09:
Well none of them will work with the current target definition, since they are not defined to be available as pins. You will have to build your own version of the firmware for this to work.

mikeller, 2017-07-09:
Hehe, overlapping posts. :-)

closed 2017-08-05 by jflyper