betaflight/betaflight #3484: KIWIF4V2 - softserial

DeepnineRC, 2017-07-09:
I'm using the KIWIF4V2.

It has some Pins I'm not using:
LED Pin  at PA0
VTX.DTA at PB7
VTX.CLK at PB6

I tired to enable softserial for my tbs unify but i can't get it working. 

RESOURCE SERIAL_TX 11 A00

LED pin does show something on the oscilloscope but it's not working.

PB7 and PB6 is not mentioned in the target files.

Mybe someone can take a look at the target.c file. I don't understand the stuff with the timers and channels.

jflyper, 2017-07-10:
@DeepnineRC 
Your LED pad is PB7.

Try

```
resource serial_tx 11 b7
```
to use LED pad for connection to unify.

DeepnineRC, 2017-07-10:
led featureis off.

jflyper, 2017-07-10:
Can you measure the supply voltage for the LED?

DeepnineRC, 2017-07-10:
There is no led voltage pad, but the 5V pads around are at 4,8V. The unify hv is running directly from battery.

jflyper, 2017-07-10:
Hmm... if PA0 doesn't work, there is nothing you can do.

May be we better summon @flyinglemonfpv

flyinglemonfpv, 2017-07-10:
In target there is no definition for softserial so any resource moving will not working. 
@DeepnineRC 
Please mail to me, I will do something for you.

DeepnineRC, 2017-07-10:
I got it working!!!! Took me just 4 hours :-)

I added timers in target.c:
    DEF_TIM(TIM4,  CH2, PB7,  TIM_USE_MOTOR | TIM_USE_LED,   1, 0),  // VTX.DTA
    DEF_TIM(TIM4,  CH1, PB6,  TIM_USE_MOTOR | TIM_USE_LED,   1, 0),  // VTX.CLK

And  TIM_N(4) to target.h

Now resource serial_tx 11 b06 and resource serial_tx 11 b07 are working with smartaudio.

@flyinglemonfpv 
Thank you for your offer, maybe you can add it to your target files (i a better way, i don't know exactly what I'm doing.)
What about the tx of the hardware uart from sbus, can this be used as softserial port too? Or the tx from ESC_TELEMTRY?

jflyper, 2017-07-10:
@DeepnineRC Yes. Unused side of hardware UARTs can be used as timer pin, provided it is associated with a valid pin and the timer is not occupied with some other function.

DeepnineRC, 2017-07-10:
Sorry! Yes i can build my own stuff, but i can't get git working. Not more than git pull for new stuff... :-)

How can i check if a timer is used by something else?

jflyper, 2017-07-10:
> How can i check if a timer is used by something else?

Ground rule is that if one of channels of a timer is used for some function, then other channels can only be used for the same function.

For example, the current KIWIF4, `timerHardware` is

```
const timerHardware_t timerHardware[USABLE_TIMER_CHANNEL_COUNT] = {
    DEF_TIM(TIM2,  CH4, PA3,  TIM_USE_MOTOR, 1, 1),
    DEF_TIM(TIM3,  CH3, PB0,  TIM_USE_MOTOR, 1, 0),
    DEF_TIM(TIM3,  CH4, PB1,  TIM_USE_MOTOR, 1, 0),
    DEF_TIM(TIM2,  CH3, PA2,  TIM_USE_MOTOR, 1, 0),
#if defined(PLUMF4) || defined(KIWIF4V2)
    DEF_TIM(TIM2,  CH1, PA0,  TIM_USE_LED,   1, 0),  //LED
#else
    DEF_TIM(TIM4,  CH2, PB7,  TIM_USE_LED,   0, 0),  // LED
#endif
};
```

Then, suppose you are using quad mixer, which use four timer channels marked `TIM_USE_MOTOR`. `TIM2` and `TIM3` are used for motors, and no other function can use these timers.

And quite interestingly, `PLUMF4` and `KIWIF4V2` case specifies `TIM2` for LED_STRIP, which actually conflicts with motor usages and I'm presuming LED_STRIPs on these boards doesn't work.

More interestingly, this is the reason why PA0 didn't work with software serial :sob:

Your new definition
```
DEF_TIM(TIM4, CH2, PB7, TIM_USE_MOTOR | TIM_USE_LED, 1, 0), // VTX.DTA
DEF_TIM(TIM4, CH1, PB6, TIM_USE_MOTOR | TIM_USE_LED, 1, 0), // VTX.CLK
```
Does not have any conflict with exisiting entries, so they are okay, unless flying a HEXA.

closed 2018-04-21 by DieHertz