betaflight/betaflight #3485: Allow BFOSD/Lua script to configure LED color

PiManASM, 2017-07-10:
TBS has a formula to convert video frequency into HSV colors, it would be neat if the FC could pass that along to addressable LEDs, and give visual feedback for your channel

mikeller, 2017-07-10:
You're too late, this has already been done: #3250 :-P

PiManASM, 2017-07-10:
Is that the TBS colorspace? I have a particular application in mind.

mikeller, 2017-07-10:
It seems to be the TBS colour space, since #3228 is referenced in #3250. What is it that you have in mind?

sheaivey, 2017-08-19:
@mikeller I'm working on this ticket #3602 It will add control to the LED Strip for overriding the current color.

> First we would need to modify the LED_Strip to accept two commands **reset()** and **setHSV()**.
> **Reset()**: removes any override color and defaults back to user configurator settings.
> **setHSV(h, s, v)**: overrides the HSV color that would have been sent to the LED Strip.

These functions will be the bases for override control of the LED Strip from OSD, Lua scripts, receiver commands, etc...

closed 2018-04-21 by DieHertz