betaflight/betaflight #3490: Little Bee 20A (original) motor twitch in 3.2

slowthisbirddown, 2017-07-11:
Oneshot or Multishot I can't get motors to stop twitching. Even after calibration. Revert to 3.1.7 and it's perfectly fine. I use a matek board and he recreated the issue himself. Please advise. Thank you.

McGiverGim, 2017-07-11:
I've similar problema in Omnibus F4 Pro with Little Bee 20A for blheli_s (I don't know if original or not). 

My problem is worst, it never worked in OneShot or D-Shot, only works with multishot, but this is surely another different problem.

jflyper, 2017-07-11:
My LB20 doesn't even respond to protocols other than OS125 😩

slowthisbirddown, 2017-07-11:
Has to be related. Hope the devs see this.

slowthisbirddown, 2017-07-13:
Fixed the issue by lowering the idle throttle command to 800. Anything above and motors will always twitch

mikeller, 2017-07-13:
Reopening this, because artificially lowering the idle throttle is a workaround, and there still is a bug causing this, and we want to keep track of this.

jflyper, 2017-07-14:
@blckmn It's #3365.

richard-scott, 2017-07-14:
I think I also get this issue too, but didn't think it was a bug.

I get this because my RX reports radio jitter onto the FC for my throttle signal... The single value it is reporting is never still at low throttle (other stick positions its fine). At low throttle the signal jumps about between 997 to 1003... When it jumps above 1000 the ESC thinks the motor should move.

I didn't think that was a bug, but a "feature" of my old DX7 radio :-)

hydra, 2017-07-18:
is this fixed in master now?

blckmn, 2017-07-18:
Sure is

closed 2018-04-21 by DieHertz