betaflight/betaflight #3493: KIWIF4V2 - LED_STRIP

maverikou, 2017-07-11:
The setup:
EMAX Bullet 30A ESCs
KIWI F4 v2
Nothing else connected.

When LED_STRIP is enabled and DSHOT is selected, motors 1 and 4 do not spin.

DMA streams do not overlap:
```
Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: MOTOR 4
DMA1 Stream 2: MOTOR 3
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: LED_STRIP
DMA1 Stream 6: MOTOR 1
DMA1 Stream 7: MOTOR 2
```

jflyper, 2017-07-11:
https://github.com/betaflight/betaflight/issues/3484#issuecomment-314168506

> And quite interestingly, PLUMF4 and KIWIF4V2 case specifies TIM2 for LED_STRIP, which actually conflicts with motor usages and I'm presuming LED_STRIPs on these boards doesn't work.

maverikou, 2017-07-11:
@jflyper So the question is, why is LED on TIM2 in the first place? @flyinglemonfpv?

jflyper, 2017-07-11:
@maverikou And I further noticed that PA0 can not be assigned to TIM2_CH1. This while line is a hoax 😢

mikeller, 2017-07-11:
I guess typo. Should probably be TIM5_CH1.

jflyper, 2017-07-11:
@mikeller I'm suspecting it and trying to make a corrected firmware, but there is something else I feel fishy about this. DMA stream/channel allocation is really messed up.

@maverikou Can you tell us which version of firmware you are running?
And, can I see outputs from "resource" and "resource list"?

mikeller, 2017-07-11:
@jflyper: What is messed up about it?

jflyper, 2017-07-11:
@mikeller Timer/channel and DMA stream doesn't match for motor 4.

jflyper, 2017-07-11:
47f4d81 PLUMF4 was added. PA0 as TIM5_CH1.
f8ea32d "KIWI V2 LED issue", KIWIF4V2 was added. PA0 still TIM5_CH1.
529b89c "Fixed KIWIF4 DSHOT DMA conflict". PA0 changed to TIM2_CH1.

mikeller, 2017-07-11:
Yeah, there's been a bunch of change for this board to make Dshot work.

jflyper, 2017-07-11:
Moving M2 and M3 to TIM1 or TIM8 and keep LED on TIM5 would be the solution...

maverikou, 2017-07-11:
This is with LED_STRIP off:

Betaflight / KIWIF4V2 3.2.0 Jul 10 2017 / 15:06:30 (491ecba)

resource BEEPER 1 A08
resource MOTOR 1 A03
resource MOTOR 2 B00
resource MOTOR 3 B01
resource MOTOR 4 A02
resource LED_STRIP 1 A00
resource SERIAL_TX 1 A09
resource SERIAL_TX 3 B10
resource SERIAL_TX 6 C06
resource SERIAL_RX 1 A10
resource SERIAL_RX 3 B11
resource SERIAL_RX 6 C07
resource INVERTER 1 C00
resource LED 1 B04
resource SPI_SCK 1 A05
resource SPI_SCK 2 B13
resource SPI_SCK 3 C10
resource SPI_MISO 1 A06
resource SPI_MISO 2 B14
resource SPI_MISO 3 C11
resource SPI_MOSI 1 A07
resource SPI_MOSI 2 B15
resource SPI_MOSI 3 C12
resource ESCSERIAL 1 A03

resource list:
A00: FREE
A01: FREE
A02: MOTOR 4
A03: MOTOR 1
A04: MPU_CS
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: BEEPER
A09: FREE
A10: SERIAL_RX 1
A11: USB
A12: USB
A13: FREE
A14: FREE
A15: OSD_CS
B00: MOTOR 2
B01: MOTOR 3
B02: FREE
B03: FREE
B04: LED 1
B05: FREE
B06: FREE
B07: FREE
B08: FREE
B09: FREE
B10: FREE
B11: FREE
B12: SPI_PREINIT
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: INVERTER 1
C01: ADC_BATT
C02: FREE
C03: FREE
C04: MPU_EXTI
C05: FREE
C06: FREE
C07: FREE
C08: FREE
C09: FREE
C10: SPI_SCK 3
C11: SPI_MISO 3
C12: SPI_MOSI 3
C13: FREE
C14: FREE
C15: FREE
D02: FREE

Currently active DMA:
DMA1 Stream 0: FREE
DMA1 Stream 1: MOTOR 4
DMA1 Stream 2: MOTOR 3
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: MOTOR 1
DMA1 Stream 7: MOTOR 2
DMA2 Stream 0: FREE
DMA2 Stream 1: FREE
DMA2 Stream 2: FREE
DMA2 Stream 3: FREE
DMA2 Stream 4: ADC
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE

jflyper, 2017-07-11:
@maverikou Can you try this?

[betaflight_3.2.0_KIWIF4V2-new_timer_mapping.hex.zip](https://github.com/betaflight/betaflight/files/1138563/betaflight_3.2.0_KIWIF4V2-new_timer_mapping.hex.zip)

Current `timerHardware` is:
```
const timerHardware_t timerHardware[USABLE_TIMER_CHANNEL_COUNT] = {
    DEF_TIM(TIM2,  CH4, PA3,  TIM_USE_MOTOR, 0, 1),
    DEF_TIM(TIM1,  CH2N,PB0,  TIM_USE_MOTOR, TIMER_OUTPUT_INVERTED, 0),
    DEF_TIM(TIM1,  CH3N,PB1,  TIM_USE_MOTOR, TIMER_OUTPUT_INVERTED, 0),
    DEF_TIM(TIM2,  CH3, PA2,  TIM_USE_MOTOR, 0, 0),
#if defined(PLUMF4) || defined(KIWIF4V2)
    DEF_TIM(TIM5,  CH1, PA0,  TIM_USE_LED,   0, 0), // LED
    DEF_TIM(TIM4,  CH1, PB6,  TIM_USE_NONE,  0, 0), // VTX.CLK
    DEF_TIM(TIM4,  CH2, PB7,  TIM_USE_NONE,  0, 0), // VTX.DTA
#else
    DEF_TIM(TIM4,  CH1, PB6,  TIM_USE_NONE,  0, 0), // VTX.CLK
    DEF_TIM(TIM4,  CH2, PB7,  TIM_USE_LED,   0, 0), // LED (VTX.CLK)
#endif
};
```

maverikou, 2017-07-13:
@jflyper Need a bit more time to fix the quad before testing the new mapping. Hopefully tomorrow.

maverikou, 2017-07-13:
FWIW here's the resource output of the 2nd firmware you posted.
Motors 2 and 3 don't work at all.

```
# resource list
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: FREE
A02: MOTOR 4
A03: MOTOR 1
A04: MPU_CS
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: BEEPER
A09: FREE
A10: SERIAL_RX 1
A11: USB
A12: USB
A13: FREE
A14: FREE
A15: OSD_CS
B00: MOTOR 2
B01: MOTOR 3
B02: FREE
B03: FREE
B04: LED 1
B05: FREE
B06: FREE
B07: FREE
B08: FREE
B09: FREE
B10: SERIAL_TX 3
B11: FREE
B12: SPI_PREINIT
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: INVERTER 1
C01: ADC_BATT
C02: FREE
C03: ADC_CURR
C04: MPU_EXTI
C05: FREE
C06: FREE
C07: FREE
C08: FREE
C09: FREE
C10: SPI_SCK 3
C11: SPI_MISO 3
C12: SPI_MOSI 3
C13: FREE
C14: FREE
C15: FREE
D02: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: MOTOR 4
DMA1 Stream 2: FREE
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: MOTOR 1
DMA1 Stream 7: FREE
DMA2 Stream 0: FREE
DMA2 Stream 1: FREE
DMA2 Stream 2: FREE
DMA2 Stream 3: FREE
DMA2 Stream 4: ADC
DMA2 Stream 5: FREE
DMA2 Stream 6: MOTOR 3
DMA2 Stream 7: FREE

Use: 'resource' to see how to change resources.

# resource
resource BEEPER 1 A08
resource MOTOR 1 A03
resource MOTOR 2 B00
resource MOTOR 3 B01
resource MOTOR 4 A02
resource LED_STRIP 1 A00
resource SERIAL_TX 1 A09
resource SERIAL_TX 3 B10
resource SERIAL_TX 6 C06
resource SERIAL_RX 1 A10
resource SERIAL_RX 3 B11
resource SERIAL_RX 6 C07
resource INVERTER 1 C00
resource LED 1 B04
resource SPI_SCK 1 A05
resource SPI_SCK 2 B13
resource SPI_SCK 3 C10
resource SPI_MISO 1 A06
resource SPI_MISO 2 B14
resource SPI_MISO 3 C11
resource SPI_MOSI 1 A07
resource SPI_MOSI 2 B15
resource SPI_MOSI 3 C12
resource ESCSERIAL 1 A03
```

jflyper, 2017-07-14:
@maverikou 
Can you try the non-dshot protocol?

@blckmn 
I wonder why motor 2 = TIM1_CH2(N) is not using DMA2_ST2 ...

Oh, I will push a PR for this, too.

closed 2017-07-24 by jflyper