betaflight/betaflight #3496: wiki: DSHOT calibration

fishpepper, 2017-07-11:
I am always reading references on how important it is to calibrate dshot escs. 

The wiki has this text in it:
> Even though Dshot doesn't use the min/max values in the ESC (BLHeli) it is recommended to do an ESC cal with one of the Analog Protocols (OneShot or MultiShot) on new and newly Flashed ESCs. This is easy to do with the BLHeli Suite's Motor Tab.

Why is that? It's useless. The only thing I could possibly imagine this to help is when you change to oneshot mode by accident. So at least it should be added that you DO NOT calibrate dshot this way and that this is just a security measure in case something/you switch back to oneshot... 

People refer to this all the time as "no, dshot needs calibration as well"... 
No way to argument against this as it is written in the holy betaflight wiki bible ;)

I vote for removal of this text.

mikeller, 2017-07-11:
I think you have to see this text in context. The sentence immediately before it reads:

> This means that when DSHOT is use NO ESC Calibration is required. Just Select DSHOT.

This makes it implicitly clear that this is just a precaution for the case when the protocol is unintentionally switches to OneShot (e.g. config reset).

I vote for clarifying the statement by stating the reason.

fishpepper, 2017-07-12:
I know.. But the web is full of comments that a calibration will affect dshot as well :-\

So what about this:

> **This means that when DSHOT is use NO ESC Calibration is required. Just Select DSHOT.**
>
>Even though DSHOT doesn't use the min/max values in the ESC (BLHeli) it is recommended to do an ESC calibration with one of the Analog Protocols (OneShot or MultiShot) on new and newly Flashed ESCs. This is easy to do with the BLHeli Suite's Motor Tab. The calibration has no effect as long as you use DSHOT. It is just a precaution in case you unintentionally switch to an old analog protocol (e.g. by config reset).

mikeller, 2017-07-12:
Compromise:

> **This means that when DSHOT is use NO ESC Calibration is required. Just Select DSHOT.**

> Even though DSHOT doesn't use the min/max values in the ESC (BLHeli) **it is recommended** to do an ESC calibration with one of the Analog Protocols (OneShot or MultiShot) on new and newly Flashed [...]

brycedjohnson, 2017-07-12:
I think it originally got into the wiki because of a blheli_s dshot early release had a bug in it that accidentally used the calibrated analog values as an offset.   Old bug workarounds tend to linger a long time even after it was fixed...

mikeller, 2017-07-13:
Amended emphasis.

closed 2017-07-13 by mikeller