betaflight/betaflight #3498: Betaflight F3 OSD

jt6064, 2017-07-12:
When trying to access on-screen menu from new Betaflight F3 running 3.1.7 with the stick commands I do not get the menu to appear.  Instead, the PID profile changes.  The buzzer sounds each time I enter the stick command (1 time for profile 1, 2 times for profile 2, etc...).  The board is working perfectly otherwise.  I am able to configure the OSD to my liking in the configurator.  I get solid video and the quad flies perfectly.  I can see the OSD through my goggles just fine.  My only issue is being unable to access the on screen menu to make setting changes or adjust my VTX.  The board is new, I've only had it 2 days.

jflyper, 2017-07-12:
What is your stick mode?

jt6064, 2017-07-12:
I fly mode 2.  I go to mid throttle, yaw left and pitch forward.  The same commands work on all my other quads running MinimOSD with the exception of the yaw direction to enter the menu.

jflyper, 2017-07-12:
@jt6064 
Profile switching all have different combinations.

https://github.com/cleanflight/cleanflight/blob/master/docs/Controls.md

If your CMS invocation stick combination really switch profiles in the way you described, you have something really weird going on in your board.

jt6064, 2017-07-12:
You're absolutely right about that.  I've got Gremlins living rent free in my board!  I need someone to help me get them out...

jflyper, 2017-07-12:
@jt6064 Which version of BF are you running atm?

jt6064, 2017-07-12:
I'm running 3.1.7.   Based on the the published stick commands, it looks like the board isn't seeing mid throttle for some reason.  The stick command to change to PID profile 2 is low throttle, yaw left and pitch forward.  This is very close to the OSD menu of throttle middle.  I wonder why it's not seeing mid throttle?  I know throttle values aren't fed to the board when disarmed not matter the throttle stick position.

AndersHoglund, 2017-07-12:
You might have throttle cut in your Tx when disarmed. I usually have that for double safety, but of course that did not work with CMS on BFF3. Had to remove throttle cut in Tx so FC sees mid throttle even when disarmed.

jt6064, 2017-07-12:
Great idea!  I never considered that.  I'll try that and report back shortly.  Thank you.

jt6064, 2017-07-12:
You are a genius, Anders!  That was the problem.  Throttle cost was inhibited and now everything works like it should.  Thank you, sir.

closed 2017-07-12 by jt6064

jflyper, 2017-07-13:
Still have ton of stuff to learn!

Happy props!