betaflight/betaflight #3507: BFF3 Softserial timer collision with motors and led strip

jflyper, 2017-07-13:
There are problems with the current timer channel assignment.
1. Enabling SOFTSERIAL1 (PB1, TX) collides with motors 3 and 4 on TIM8.
2. Enabling SOFTSERIAL1 (PB0, RX) collides with led strip on TIM1.

I would suggest using TIM3_CH3 (PB0) and TIM3_CH4 (PB1) instead.
Thoughts?

Ref:
https://github.com/betaflight/betaflight/blob/master/src/main/target/BETAFLIGHTF3/target.h#L71-L72
```
 71 #define SOFTSERIAL1_RX_PIN      PB0 // PWM 5
 72 #define SOFTSERIAL1_TX_PIN      PB1 // PWM 6
```

https://github.com/betaflight/betaflight/blob/master/src/main/target/BETAFLIGHTF3/target.c#L28-L39
```
28 const timerHardware_t timerHardware[USABLE_TIMER_CHANNEL_COUNT] = {
 29     DEF_TIM(TIM4, CH2, PB7, TIM_USE_PPM,                 TIMER_INPUT_ENABLED), // PPM IN
 30     DEF_TIM(TIM16,CH1, PA6, TIM_USE_MOTOR,               TIMER_OUTPUT_ENABLED), // PWM1
 31     DEF_TIM(TIM8, CH1N,PA7, TIM_USE_MOTOR,               TIMER_OUTPUT_ENABLED | TIMER_OUTPUT_INVERTED), // PWM2
 32     DEF_TIM(TIM8, CH2, PB8, TIM_USE_MOTOR,               TIMER_OUTPUT_ENABLED), // PWM3
 33     DEF_TIM(TIM17,CH1, PB9, TIM_USE_MOTOR,               TIMER_OUTPUT_ENABLED), // PWM4
 34     DEF_TIM(TIM1, CH2N,PB0, TIM_USE_MOTOR,               TIMER_OUTPUT_ENABLED | TIMER_OUTPUT_INVERTED), // PWM5
 35     DEF_TIM(TIM8, CH3N,PB1, TIM_USE_MOTOR,               TIMER_OUTPUT_ENABLED | TIMER_OUTPUT_INVERTED), // PWM6
 36     DEF_TIM(TIM2, CH1, PA0, TIM_USE_MOTOR,               TIMER_OUTPUT_ENABLED), // PWM7
 37     DEF_TIM(TIM2, CH3, PA2, TIM_USE_MOTOR,               TIMER_OUTPUT_ENABLED), // PWM8
 38     DEF_TIM(TIM1, CH1, PA8, TIM_USE_MOTOR | TIM_USE_LED, TIMER_OUTPUT_ENABLED), // LED_STRIP --requires resource remap w    ith dshot (remap to motor 5??)--
 39 };
```

borisbstyle, 2017-07-13:
I believe this is done to support dshot on m5 and m6. I cant remember exactly, but you can give it a try

jflyper, 2017-07-13:
Okay, the `timerHardware` is further broken:

3. PWM7 and 8 was (iirc) intended as aliases of UART2_{RX,TX}, in which case they should be PA3 and PA2.

jflyper, 2017-07-13:
@borisbstyle
> I believe this is done to support dshot on m5 and m6. I cant remember exactly, but you can give it a try

IC...

jflyper, 2017-07-16:
#3535 finalize the change.

closed 2017-07-16 by jflyper