betaflight/betaflight #3508: BF 3.2 - PID profiles show wrong profile. doesnt save every time.

riquezjp, 2017-07-13:
If I go into the OSD & select PID 3 & look at the PID, then go back & change to PID 2 & then look at PID 2 it will say PID 2 but it will show me PID3
If I go to #1 first & then look at #2, it will show me #1 again.

If you make changes to the PID & select "save & reboot" it may or may not save. I havent worked out why it dosnt always save. it maybe to do with moving the sticks during the reboot  - cant be certain.

markgibbons25, 2017-07-13:
I have this issue too but since I haven't tried OSD on 3.1.x I wasn't sure if it was a bug or not. I think the work around is that you must always use roll right to save and get back to the main menu and then roll right to save & exit.

riquezjp, 2017-07-13:
I do not have these issue in 3.1.6, but in 3.2 there are a lot of new options. & of course, it's still in dev...

There are also a couple of other OSD glitches i've noticed, but haven't had chance to catch them for sure yet.
Your last flown PID seems to change sometimes. 
For example if I fly PID 3, then change battery, i might then be flying PID 1 next time :-o

I also think sometimes when changing PID3, it actually changes PID2 or vice/versa - this is probably related to the main issue of it displaying the wrong profile.

mikeller, 2017-07-13:
I think there was a bug with stick commands being recognised while OSD was active. Could this be a case of your OSD PID changes being interpreted as profile switching commands?
As far as I can remember, this bug was fixed. Can you try again with latest 3.2.?

riquezjp, 2017-07-13:
Watching the DVR video you can see that the page says "Profile 2" but its showing the PID for profile 3 (or profile 1), which ever you last viewed. 

So, just to be clear - 
• enter OSD & view Profile 2 (Roll P is 44)
• back & view Profile 3 (Roll P is 45)
• back & view Profile 2 (it says Profile 2, but the PID displayed is Profile 3)

DanNixon, 2017-07-14:
Most of the issues you mention are fixed in the referenced PR.

I've been unable to reproduce the PID profile being different when power cycling. The only case where this will definitely happen is if you select "Exit" on the main menu (rather than either "Save" option) or unplug the battery while still in the CMS.

riquezjp, 2017-07-14:
OK great Dan! Thanks.
I look forward to catching up soon, been working on my tune... then will get the next bus to BF3.2 😉

martinbudden, 2017-07-15:
Closing, fixed in PR https://github.com/betaflight/betaflight/pull/3518

PR #3518 (2017-07-15): Ensure PID/rate profile is updated before editing values in CMS
Ensure the PID and rate profiles are set as soon as they are changed, ensures that the values being edited are for the correct selected profile.

Fixes #3508
closed 2017-07-15 by martinbudden