betaflight/betaflight #3512: Building SITL as target on x86 desktop requires ARM toolchain?

ohshout, 2017-07-13:
Hello! I'm new to betaflight/cleanflight. I followed this tutorial: SITL in gazebo 8 with ArduCopterPlugin and it gave me an error:
make: arm-none-eabi-gcc: Command not found
make/tools.mk:294: *** **ERROR** arm-none-eabi-gcc not in the PATH. Run 'make arm_sdk_install' to install automatically in the tools folder of this repo.  Stop.
 
I understand that the native firmware runs on ARM, but why the SITL requires ARM toolchain, too? It is just a linux process on x86, isn't it? Have I misunderstood anything?

mikeller, 2017-07-13:
No, this is indeed a bug in the makefile, this check should be suppressed for the SITL target.

For now, just comment out this line to get it going:
https://github.com/betaflight/betaflight/blob/master/Makefile#L82

ohshout, 2017-07-13:
@mikeller Thanks, that works! Is the patch just as simple as:
  81 # include the tools makefile
  82 ifeq ($(TARGET),$(filter $(TARGET), $(SITL_TARGETS)))
  83   include $(ROOT)/make/tools.mk
  84 endif

I've tested it. It works. Should I submit the patch?

DanNixon, 2017-07-15:
This also applies to the unit tests too. That patch would also cover that case as the `TARGET` variable would be (or at least should be) an empty string when running the tests.

blckmn, 2017-07-22:
I think this is now resolved. Can you please confirm - and if not then reopen?

closed 2017-07-22 by blckmn