betaflight/betaflight #3513: AnyFC F7 - external sensor and I2C not working

ThomasH3, 2017-07-13:
Hi 
I've connected a magnetometer and a PDB with VBAT and current sensor to my AnyFc F7. 
It seems I2C and ADC Input (vbat and current) are not recognised. the wiring  is correct.
Is there anything else then the part of the "configuration" tab which I need to enable ?

jflyper, 2017-07-14:
@ThomasH3 
Can we see results of CLI commands "resource" and "resource list"?

ThomasH3, 2017-07-14:
A00: MOTOR 4
A01: MOTOR 1
A02: SERVO 2
A03: MOTOR 2
A04: MPU_CS 
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A11: USB 
A12: USB  
B05: MOTOR 3
B06: LED 2
B07: LED 1
B08: SERVO 1
C00: ADC_BATT 
C01: ADC_CURR 
C04: MPU_EXTI 
C10: SERIAL_TX 4
C12: SERIAL_TX 5
D02: SERIAL_RX 5
D06: SERIAL_RX 2
D12: I2C_SCL 4
D13: I2C_SDA 4
E07: SERIAL_RX 7
E08: SERIAL_TX 7
E11: SPI_CS 4
E12: SPI_SCK 4
E13: SPI_MISO 4
E14: SPI_MOSI 4

Currently active DMA:
DMA1 Stream 1: SERIAL_TX 7
DMA1 Stream 4: SERIAL_TX 4
DMA1 Stream 6: SERIAL_TX 2
DMA1 Stream 7: SERIAL_TX 5
DMA2 Stream 4: ADC

jflyper, 2017-07-14:
Oh... I see what's going on with I2C.

The firmware enables I2C on I2C4, which is connected to (non-existent?) MS5611.
As I look at the schematics, I2C pad/TH is I2C2, and that is not enabled.

@ThomasH3 Two questions.
Do you have MS5611 installed?
What kind of external sensor(s) are you connecting?

ThomasH3, 2017-07-14:
No barometer installed. I'm trying to connect a neo-7m gps module with compass. Gps is working

jflyper, 2017-07-14:
@ThomasH3 
We are moving to #3515.

DzikuVx, 2017-07-14:
@jflyper in INAV we "fixed" this problem by creating a temporary target for those banggood clones without baro BTW

jflyper, 2017-07-14:
@DzikuVx Gees, I never thought there were ones with baro installed!!! Are they common?

DzikuVx, 2017-07-14:
@jflyper no, not so much. But @sambas has one for sure and also I know about one other case. Probably the case without baro is majority of the boards

jflyper, 2017-07-14:
We can deal with variations as we proceed with "configurable XXX" :)

ThomasH3, 2017-07-14:
Inav fw upload is not working I can only load cleanflight and betaflight. Do you have any idea why external current is not been detected?

wind0r, 2017-07-14:
Because banggood cloned a board that never was released and never official supported :-)

jflyper, 2017-07-16:
#3515 
Activated I2C2 and made it default.
Onboard I2C4 and associated baro (MS5611) will become accessible again when #3530 is finished.

@ThomasH3 Can you close this?

DzikuVx, 2017-07-16:
@ThomasH3 you might want to open an issue in INAV project about being unable to flash AnyFC F7.
https://github.com/iNavFlight/inav
I was flashing mine few days ago so I suppose there is something fishy going on there

closed 2017-07-16 by ThomasH3