betaflight/betaflight #3516: CLI command 'resource'  problem

LuctenFinger, 2017-07-14:
Once upon a time , i use CLI command resource to remap my motor, it works well.

Now I want to remap UART 3 , but it fail .

Is it not impossible to remap UART ? or it is work-in-progress?

My board is Omnibus F4 AIO V1.

# RESOURCE
resource BEEPER 1 B04
resource MOTOR 1 B01
resource MOTOR 2 B00
resource MOTOR 3 A03
resource MOTOR 4 A02
resource PPM 1 B14
resource PWM 1 B14
resource PWM 2 B15
resource PWM 3 C06
resource PWM 4 C07
resource PWM 5 C08
resource PWM 6 C09
_**resource SERIAL_TX 3 A01
resource SERIAL_RX 3 A08**_

# RESOURCE LIST
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE 
A01: FREE 
A02: MOTOR 4
A03: MOTOR 3
A04: MPU_CS 
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: FREE 
A09: FREE 
A10: SERIAL_RX 1
A11: USB 
A12: USB 
A13: FREE 
A14: FREE 
A15: OSD_CS 
B00: MOTOR 2
B01: MOTOR 1
B02: FREE 
B03: FLASH_CS 
B04: BEEPER 
B05: LED 1
B06: FREE 
B07: FREE 
B08: FREE 
B09: FREE 
**_B10: SERIAL_TX 3
B11: SERIAL_RX 3_**
B12: FREE 
B13: FREE 
B14: FREE 
B15: FREE 
C00: FREE 
C01: FREE 
C02: ADC_BATT 
C03: FREE 
C04: MPU_EXTI 
C05: FREE 
C06: FREE 
C07: FREE 
C08: FREE 
C09: FREE 
C10: SPI_SCK 3
C11: SPI_MISO 3
C12: SPI_MOSI 3
C13: FREE 
C14: FREE 
C15: FREE 
D00: FREE 
D01: FREE 
D02: FREE 
D03: FREE 
D04: FREE 
D05: FREE 
D06: FREE 
D07: FREE 
D08: FREE 
D09: FREE 
D10: FREE 
D11: FREE 
D12: FREE 
D13: FREE 
D14: FREE 
D15: FREE 

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: FREE
DMA2 Stream 1: FREE
DMA2 Stream 2: FREE
DMA2 Stream 3: FREE
DMA2 Stream 4: ADC
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE

Arakon, 2017-07-14:
You can't reassign hardware serial to different pins, there's actual hardware on the chip for these ports that can't just be moved.

LuctenFinger, 2017-07-14:
thanks anyway

LuctenFinger, 2017-07-15:
I have several sensor such as GPS, baro, sonar, but there s only UART 1 , UART3, UART6 on my board 

but UART 1 is use by SBUS, UART3 is use by i2c, so I have only 1 UART port to use.

this is despair.  is there any way to use these sensor all together?

Arakon, 2017-07-15:
Softserial will work for GPS, for example.

LuctenFinger, 2017-07-15:
thx a lot !

closed 2017-07-15 by LuctenFinger