betaflight/betaflight #3525: [Feature Request] Allow for more than 4 AUX channels on MSP_SET_RAW_RC

seeul8er, 2017-07-14:
**Even cheap 50€ RCs have up to 6 AUX channels. With the increasing number of configuration options more AUX channels come in handy.** 
As it is a digital serial protocol more channels are not as problematic as with analog or PWM radio controls. Make better use of the benefits offered by MSP_RC compared to traditional RCs.
MSP_SET_RAW_RC is very useful in combination with external hardware (like Raspberry Pis) and custom/software receivers (e.g. wifibroadcast). A camera gimbal alone uses half of the four available channels. 

Options:

1. automatically detect number of channels sent over MSP
2. CLI/MSP variable to define number of MSP AUX channels (fewer iterations. faster?)

jflyper, 2017-07-15:
@seeul8er It looks like the `MSP_SET_RAW_RC` does accept up to `MAX_SUPPORTED_RC_CHANNEL_COUNT` already. Did you give it a try?

seeul8er, 2017-07-15:
Just tested it and it does 5 AUX channels over MSP without a problem. So I guess this is solved :grinning: 
Read something about a max of 4 channels because of legacy support or something some while ago. Should have checked it in the first place!

Thank you!

closed 2017-07-15 by seeul8er