betaflight/betaflight #3542: Motors stopping briefly on punch outs

jsweeney91, 2017-07-16:
During punch outs the motors temporarily stop spinning. Beeper isn't signalling RX_LOSS or anything so I don't think it's failsafe, had replies from people with the same situation:
https://www.facebook.com/groups/rotorriot/permalink/859664410855199/?comment_id=859666047521702&reply_comment_id=859675447520762&notif_t=group_comment_reply&notif_id=1500210247853175
and somebody else has a video with the same thing too(35 seconds-ish):
https://www.youtube.com/watch?v=N8VQatepBqs 

TBS powercube running 3.1.7 airmode and anti-gravity are both active

Arakon, 2017-07-16:
The group is closed, no way to see the video.
But with the second video, that's a classic desync. Considering hundreds of people don't have this issue, it's likely to be a hardware problem/hardware combination issue or wrong settings.

jsweeney91, 2017-07-17:
@Arakon would a desync not cause a roll of death? the only other hardware apart from the powercube that's relevant would be the motors

jsweeney91, 2017-07-17:
and how likely is it that all 4 motors would desync simultaneously ?

jsweeney91, 2017-07-17:
@Arakon 2:11 of this video:
https://youtu.be/8BiBlWJV-qA

Arakon, 2017-07-17:
Could still be failsafe.. I agree desync on all four is unlikely. Failsafe doesn't trigger unless the signal is lost for more than 0.5 seconds, a very brief loss of signal could do this. If you can, post a blackbox log of it happening.

Je47, 2017-07-17:
I experienced something similar and it was my throttle value going beyond the fail safe max check by just a couple digits( only happened when really mashing the stick up). trimmed the throttle back and it was fixed.
-check receiver values min/max and failsafe check value

jsweeney91, 2017-07-17:
@Arakon perhaps but it only happens during punchouts. I don't currently have a blackbox recorder but I'll try to get one

jsweeney91, 2017-07-17:
@Arakon confirmed not failsafe as RX is set to hold last values, tested on bench too

Cocobeasdad, 2017-07-22:
Running 3.1.7 on an F3 mantis board, 3s, stock PIDs, 5" props, 30a flycolor blheli_s running Dshot125.  Unbalanced props, cheap plastic frame, baby beast motors (Proverbial V8 motor install in a 75 Ford Pinto).  During punch outs quad would roll left and yaw counterclockwise until throttle reduced.  Incremental throttle increase would not produce this.  Went back to 3.1.6 and problem went away.  Saw this was an issue a year ago with prior versions.