betaflight/betaflight #3543: Motor test mode not working after loading 3.2.0

spiffy59, 2017-07-16:
I flashed betaflight_3.2.0_FF_FORTINIF4.  I have no responce in motor test mode.  I can select the "i agree tab" but the master or single sliders won budge.  Flashin back it 1.7 works fine

jflyper, 2017-07-17:
@spiffy59 If you are using development version of the firmware, you have to use corresponding development version of configurator.

You can close this.

spiffy59, 2017-07-17:
can i get location for the development version of betaflight? i am new to this

jflyper, 2017-07-17:
@spiffy59 
Development versions are not recommended for non-experts...

spiffy59, 2017-07-17:
wow pretty rude.   I have been coding for thirty years.  I got into this hobby to learn something new.  I have invested my own money and my own time to learn.  Where do you come off passing judgment?
I am going to ask again. where do i find find development version of beta flight?  If you dont want to help please keep quiet

jflyper, 2017-07-17:
@spiffy59 Sorry, no offense. I just wanted to say that this is not a place for support.

You can download the latest dev BFC from betaflight-configurator repo. (Go back up one level and you'll find it.)

spiffy59, 2017-07-17:
thank you.  i will keep that in mind in the future.

wind0r, 2017-07-17:
nightly releases need nightly configurator

if you click on wiki in the top bar and not issues nearly a lot of questions like pre build nightlys and stuff are answered there. this like nightlys is even on the first site. this is to keep the issue tracker clean.

btw issue tracker. could you please close the issue since it already got answered

@jflyper wasn't rude. he only was careful. with only reading this issue and not your cv no one can know you knowledge level. without reading the wiki and knowing that a nightly and not a officall release got flashed strange bugs can happen. so it is not recommended for beginners or as @jflyper  said only for experts since no one will pay for errors.

closed 2017-07-18 by spiffy59