betaflight/betaflight #3550: dshot1200 not working correctly on spracing f3

knockoutperformance, 2017-07-18:
dshot600 seems to work ok, a little noisy through motors. (sounds faintly like an old dial up modem, though)

dshot1200 does not spin mtr 4 with master control, mtr2 and mtr3 individual tabs run funny or not at all. 

remapping like the spracingf3 evo on the dshot wiki does not work either, tried many other combos, with little success. got it close, but not close enough, then dug into the dma timers and pins.

made a fork and remapped the pinout and dma timers for only the spracing f3 under SPRACING_DSHOT1200.
went down to 10 timers. lost one of the softserial uarts, but could add an vcp.
can use led, ppm with 6 motors, hexa8 will work but loses the led.
added a lighten_me if not needing the extra drivers for mag, baro, sonar.

jflyper, 2017-07-18:
@knockoutperformance 
Why do you want to run dshot1200 on F3?
I think it's way overkill for 16K pid freq.

Am I missing something?

knockoutperformance, 2017-07-18:
I really didnt want to run dshot1200, was happy with 300. But one of my old esc's died, and it was time for some upgrades. I got new esc's that can run dshot1200 and telemetry. Only had an spracingf3 deluxe to test with, wanted to make sure they worked correctly before buying an nerof7.

Glad i did check , one esc was fualty.

Yeah, overkill on the sprf3. 

With timer changes, between dshot600 and dshot1200, On my setup, it makes a difference. Less audible noise in the motors while running, they idle better, and a little less noise into the gyro from mtrs.

No, your not missing anything. Most likely, i am, with the f3.

jflyper, 2017-07-19:
@knockoutperformance Okay, I understood your intensions.
You can close this once your challenge is over 😉

closed 2017-07-19 by knockoutperformance

apurvaChaudhari, 2022-01-28:
Hi @knockoutperformance,
I also have dshot1200 esc , please share your fork ,if possible