betaflight/betaflight #3553: [Feature request] As a user I want to be able to copy profile settings to another profile to edit settings

basdelfos, 2017-07-18:
As a user I want to be able to copy profile settings to another profile to edit settings.
Please assign this issue to me, I put this on my todo list.

mikeller, 2017-07-18:
I think it would make more sense to do this in configurator, and not in the firmware.

basdelfos, 2017-07-18:
So, do not implement this in the OSD only from the configurator?

chime13, 2017-07-18:
A "copy Profile _n_ to Profile _x_" function in CLI would be easiest to implement

basdelfos, 2017-07-26:
See PR: https://github.com/betaflight/betaflight/pull/3636
And configurator PR: https://github.com/betaflight/betaflight-configurator/pull/566

chime13, 2017-07-29:
I tried Configurator 3.1.2 hoping that the function would be implemented but the new buttons don't show up....I used the last updated messages.json....

basdelfos, 2017-07-30:
@chime13 this PR and the BF PR are not merged yet, therefor it is not available yet.

jflyper, 2017-10-23:
Closed via #3636 and betaflight/betaflight-configurator#566

closed 2017-10-23 by jflyper