betaflight/betaflight #3574: BF 3.2 - OSD Menu unresponsive

krich1987, 2017-07-20:
I tried making changes to PIDs in the OSD, but it won't respond to any inputs after getting into the menu. This was on the 7/19/17 build of BF 3.2, running on a BFF3.

jflyper, 2017-07-20:
Keep your left stick dead center.

krich1987, 2017-07-20:
I don't think I was even touching the left stick (M9 gimbals, too, so I doubt it's jitter). In one case I was able to scroll down one step before it froze.

jflyper, 2017-07-20:
@krich1987 Please try it.

cycoconutz, 2017-07-20:
had the same issue, centering left stick worked for me.

krich1987, 2017-07-20:
I just realized that releasing the left stick will only center yaw. I'll try centering throttle as well next time. 

In any case, this should be changed to behave like the menu on 3.1.X.

DanNixon, 2017-07-20:
@krich1987 Requiring mid throttle for the CMS was to prevent accidental triggering of "legacy" stick commands.

jflyper, 2017-07-20:
#3580

closed 2017-08-04 by jflyper