betaflight/betaflight #3576: BF 3.2 - OMNIBUSF4SD Barometer not working

jangrewe, 2017-07-20:
Hi, i just flashed a recent build of BF 3.2 to my Omnibus F4 Pro v3, and the barometer doesn't work - it was fine in BF 3.1.7

I've tried setting `baro_bustype = spi` and `baro_hardware=bmp280` (and of course the baro is enable on the configuration page), but it is greyed out in the Sensors page and shows 0.0 in the OSD.

jflyper, 2017-07-20:
You can make it work by
```
set baro_bustype = spi
set baro_spi_device = 3
resource baro_cs b3
```
You can leave `baro_hardware` to `AUTO`.

The reason behind this phenomenon is that OMNIBUSF4SD target has both I2C and SPI variants of BMP280 defined. The choice was exclusive prior to #3530, but the PR made it possible to have both drivers to be configured, and defaults to I2C version as to give external sensors priority. This behavior might be changed during RC phase of 3.2.

jangrewe, 2017-07-20:
Thanks, it works with your configuration. Is there any chance this could be statically defined for this board (later on), as - i guess - all Omnibus F4 with SD should be identical?

jflyper, 2017-07-21:
I'm thinking about it ...

Noseflight, 2017-08-01:
hi jangrewe, I can only tell you i have the asgard v2 , and have to use also omnibusf4sd target and as far as i know i dont have a barometer on it :)

closed 2017-08-01 by jflyper

jangrewe, 2017-08-01:
@Noseflight I was using the OMNIBUSF4SD target, as you can see in my first post, but it's working now without any manual adjustment, thanks to the changes made by @jflyper.