betaflight/betaflight #3579: B.F. 3.2 Flip 32 Omnibus F7 - no sbus, dump in cli causing lockup

madchiller, 2017-07-20:
Using Flip32 Omnibus F7.    Today I tried the latest build #1484 e0a7594

So far I'm getting no response from my RX.  Unit powers and is bound.  Using RX1 on uart1.   Configurator shows no input.   Serial is enabled on uart1.

Also whenever I do a dump command in cli.  Everything just locks up.  If I click elsewhere in the configurator the screen just show,waiting for data.  I have to physically unplug the FC and reboot.

madchiller, 2017-07-20:
DUMP FILE:

2017-07-20 @ 09:07:34 -- Running - OS: Windows, Chrome: 59.0.3071.115, Configurator: 3.1.2
2017-07-20 @ 09:33:26 -- Serial port successfully opened with ID: 2
2017-07-20 @ 09:33:26 -- MultiWii API version: 1.35.0
2017-07-20 @ 09:33:26 -- Flight controller info, identifier: BTFL, version: 3.2.0
2017-07-20 @ 09:33:26 -- Running firmware released on: Apr 26 2017 22:55:35
2017-07-20 @ 09:33:26 -- Board: OBF7, version: 0
2017-07-20 @ 09:33:26 -- Unique device ID: 0x3900403136510e35333830
2017-07-20 @ 09:33:26 -- Craft name:
2017-07-20 @ 09:34:17 -- EEPROM saved
2017-07-20 @ 09:34:17 -- Device - Rebooting
2017-07-20 @ 09:34:17 -- Unrecoverable failure of serial connection, disconnecting...
2017-07-20 @ 09:34:17 -- Serial port successfully closed
2017-07-20 @ 09:34:25 -- Serial port successfully opened with ID: 3
2017-07-20 @ 09:34:25 -- MultiWii API version: 1.35.0
2017-07-20 @ 09:34:25 -- Flight controller info, identifier: BTFL, version: 3.2.0
2017-07-20 @ 09:34:25 -- Running firmware released on: Apr 26 2017 22:55:35
2017-07-20 @ 09:34:25 -- Board: OBF7, version: 0
2017-07-20 @ 09:34:25 -- Unique device ID: 0x3900403136510e35333830
2017-07-20 @ 09:34:25 -- Craft name:
2017-07-20 @ 09:35:52 -- Serial port successfully closed
2017-07-20 @ 09:35:54 -- Loaded release information from GitHub.
2017-07-20 @ 09:36:03 -- Using cached release information.
2017-07-20 @ 09:39:49 -- Failed to open serial port
2017-07-20 @ 09:39:57 -- Failed to open serial port
2017-07-20 @ 09:40:28 -- Serial port successfully opened with ID: 5
2017-07-20 @ 09:40:28 -- MultiWii API version: 1.35.0
2017-07-20 @ 09:40:28 -- Flight controller info, identifier: BTFL, version: 3.2.0
2017-07-20 @ 09:40:28 -- Running firmware released on: Apr 26 2017 22:55:35
2017-07-20 @ 09:40:28 -- Board: OBF7, version: 0
2017-07-20 @ 09:40:28 -- Unique device ID: 0x3900403136510e35333830
2017-07-20 @ 09:40:28 -- Craft name:
2017-07-20 @ 09:40:29 -- Serial port successfully closed
2017-07-20 @ 09:40:47 -- Using cached release information.
2017-07-20 @ 09:40:57 -- Failed to open serial port
2017-07-20 @ 09:41:42 -- USB device successfully opened with ID: 1
2017-07-20 @ 09:41:42 -- Detected device with total flash size 1024 kiB
2017-07-20 @ 09:41:42 -- Read protection not active
2017-07-20 @ 09:41:46 -- Erased 480 kB of flash successfully
2017-07-20 @ 09:42:04 -- USB device successfully closed
2017-07-20 @ 09:42:38 -- Serial port successfully opened with ID: 8
2017-07-20 @ 09:42:38 -- MultiWii API version: 1.36.0
2017-07-20 @ 09:42:38 -- Flight controller info, identifier: BTFL, version: 3.2.0
2017-07-20 @ 09:42:38 -- Running firmware released on: Jul 20 2017 14:10:02
2017-07-20 @ 09:42:38 -- Board: OBF7, version: 0
2017-07-20 @ 09:42:38 -- Unique device ID: 0x3900403136510e35333830
2017-07-20 @ 09:42:38 -- Craft name:
2017-07-20 @ 09:42:49 -- EEPROM saved
2017-07-20 @ 09:42:49 -- Unrecoverable failure of serial connection, disconnecting...
2017-07-20 @ 09:42:49 -- Serial port successfully closed
2017-07-20 @ 09:42:52 -- Serial port successfully opened with ID: 9
2017-07-20 @ 09:42:52 -- MultiWii API version: 1.36.0
2017-07-20 @ 09:42:52 -- Flight controller info, identifier: BTFL, version: 3.2.0
2017-07-20 @ 09:42:52 -- Running firmware released on: Jul 20 2017 14:10:02
2017-07-20 @ 09:42:52 -- Board: OBF7, version: 0
2017-07-20 @ 09:42:52 -- Unique device ID: 0x3900403136510e35333830
2017-07-20 @ 09:42:52 -- Craft name:
2017-07-20 @ 09:43:05 -- EEPROM saved
2017-07-20 @ 09:43:05 -- Unrecoverable failure of serial connection, disconnecting...
2017-07-20 @ 09:43:05 -- Serial port successfully closed
2017-07-20 @ 09:43:09 -- Serial port successfully opened with ID: 10
2017-07-20 @ 09:43:09 -- MultiWii API version: 1.36.0
2017-07-20 @ 09:43:09 -- Flight controller info, identifier: BTFL, version: 3.2.0
2017-07-20 @ 09:43:09 -- Running firmware released on: Jul 20 2017 14:10:02
2017-07-20 @ 09:43:09 -- Board: OBF7, version: 0
2017-07-20 @ 09:43:09 -- Unique device ID: 0x3900403136510e35333830
2017-07-20 @ 09:43:09 -- Craft name:
2017-07-20 @ 09:43:24 -- EEPROM saved
2017-07-20 @ 09:43:24 -- Unrecoverable failure of serial connection, disconnecting...
[... 865 lines cut ...]
set osd_tim_1_pos = 2102
set osd_tim_2_pos = 2081
set osd_flymode_pos = 2381
set osd_throttle_pos = 225
set osd_vtx_channel_pos = 2425
set osd_crosshairs = 2248
set osd_horizon_pos = 2248
set osd_current_pos = 385
set osd_mah_drawn_pos = 353
set osd_craft_name_pos = 2410
set osd_gps_speed_pos = 218
set osd_gps_lon_pos = 82
set osd_gps_lat_pos = 65
set osd_gps_sats_pos = 51
set osd_home_dir_pos = 302
set osd_home_dist_pos = 303
set osd_compass_bar_pos = 266
set osd_altitude_pos = 247
set osd_pid_roll_pos = 0
set osd_pid_pitch_pos = 0
set osd_pid_yaw_pos = 0
set osd_debug_pos = 1
set osd_power_pos = 321
set osd_pidrate_profile_pos = 345
set osd_

madchiller, 2017-07-20:
Also after flashing this board today i can no longer to seem to get it load the bootloader...

madchiller, 2017-07-20:
well,  After doing reset settings... The RX now works.  Dump in CLI now seems to reboot the FC wen I click out of CLI.

jflyper, 2017-07-20:
@madchiller Can you do CLI `defaults`?

madchiller, 2017-07-20:
CLI Dump no longer hangs, but if i click on say Ports..  It reconnects to the FC.  Still cannot get bootloader working.

jflyper, 2017-07-20:
> CLI Dump no longer hangs, but if i click on say Ports.. It reconnects to the FC.

This is a common and standard behavior for every FC.

> Still cannot get bootloader working. Was going to try to reflash and full chip erase. FC just seems to boot normally with boot pins shorted.

This has nothing to do with the firmware, as BOOT0 is sensed by MCU's internal boot loader long before BF firmware runs.

madchiller, 2017-07-20:
@jflyper   Well.  I worked earlier today and now NO LOVE after firmware update....  Im trying to zadig it, but so far no love..

madchiller, 2017-07-20:
@jflyper   Thank you for your help.. I managed to knock it into DFU mode using the Impulse Driver Fixer tool.  Did a full chip erase and things seems to be working well.  I guess this can be closed...

TY

closed 2017-07-21 by martinbudden