betaflight/betaflight #3580: Introducing modality into stick commands

jflyper, 2017-07-20:
CMS stick command detection had been strengthened since 3.1.7 so that left stick must be centered while navigating with right stick, to prevent accidental detection of existing stick commands.

Since then, there has been increasing number of complaints that CMS navigation freezes  after entering the menu, and more are expected during 3.2RC phase.

So, here is the problem statement.

**_Should we introduce modality in stick commands so that detection of existing stick commands is disabled while CMS is active?_**

DieHertz, 2017-07-20:
I think we should introduce stick command "context" or even a stack of contexts, allowing the same stick commands being used for different purposes based on which screen we're currently on.

jflyper, 2017-07-21:
@DieHertz Woo, isn't that over engineered?
I was thinking exposing `cmsInMenu` bool and referencing it at the beginning of `processRcStickPositions` to proceed or not.

arock12, 2017-07-30:
Im having this exact problem right now. Everytime i enter the osd menu it freezes. Any fixes?

jflyper, 2017-07-30:
@arock12 If you read the message immediately above yours, you should understand the fix is on the way 😉

DieHertz, 2017-07-30:
@jflyper I'm fine with it, just thinking big from the begging)

closed 2017-07-31 by jflyper