betaflight/betaflight #3583: Dropping support for GYRO chip integrated MAG

jflyper, 2017-07-21:
We currently support two gyro chip integrated mags; AK8963 (MPU9250) and AK8975 (MPU9150).

I would like to drop support for these mags toward 3.2 or 3.3 or 4.0, because current (3.2) bus device abstraction is not good at handling nested buses.

- Are there targets that depend solely on these mags?
There are 19 or so targets that currently configure AK8963 or AK8975. Are these vital for these  targets, or just added for bonus because gyro has it?

- Are these mags really useful?
It is always recommended to place mag sensors as far as possible from strong current. For this reason, onboard mag sensors are usually regarded as useless.

Thoughts?

MJ666, 2017-07-21:
As you can see there are many FC making use of the integrated mag. Even if this is not the perfect solution it will work after. In general having a mag somewhere separate will be quite useful i have seen quite some setups where an FC integrated mag is working quite well and does not show interferences. This really depends how the copter is build. I think we need to find a way to make this work with the bus device abstraction. For I2C it should not be an big issue since the mag will show u as additional device on the bus. The tricky part will be the SPI connected sensors with integrated mag. Please be aware im maintaining also #3215 since a while and have done some major updates during the last days to keep track of the ongoing changes.

martinbudden, 2017-07-21:
I'm not in favour of dropping support.
1. It's an indication that our bus abstraction needs improving. So we should improve it.
2. Integrated mags aren't totally useless - it depends on the construction of the aircraft. (But agreed, they are suboptimal compared to external mag.)
3. I've heard of gyro chips with integrated barometers (ICM-20789 https://www.invensense.com/products/motion-tracking/7-axis/). Those would be useful, so we would have to solve the bus abstraction for those devices.

jflyper, 2017-07-21:
@MJ666 @martinbudden 
Okay. I'm convinced. I'll probably extend appropriate device structure (busDevice_t or magDev_t) to include pointers to register read/readBuffer/write function pointers for the bus the device resides. Let me give it a try.

@MJ666 
Yes, I'm totally aware of #3215, and was kind of waiting for it to be merged before working on configurable compass to save you some work 😉

brucesdad13, 2017-07-21:
Both my targets use MPU-9250 and I'd prefer to keep integrated mag support awhile longer ;)

jflyper, 2017-07-24:
I have the first attempt in #3613.

closed 2017-08-05 by jflyper

fiveangle, 2017-08-05:
Anecdotally, I've found that if you have arm-mounted ESCs, the onboard mag works quite well on 220 class (130 class i have problems with) and the mag is very useful for those of us using UHF control and not dedicated to racing-only.  Thanks for trying to keep support for these sub-optimal devices.