betaflight/betaflight #3588: Crash Recovery

Vidalcris, 2017-07-21:
Hi,
I know this is still a beta feature but as i was trying it there is something i dont understand..
When i activate the feature, i'm not anymore able to do fast flips :-/
Look like the feature think that i just crashed and try to level.
With crash_gthreshold = 500 its working better but still got the problem when i combine a fast roll+pitch at the same time.
My FC is a Omnibus F3.

Vidalcris, 2017-07-21:
Mhm i'm just thinking that my accelerometer is disabled.. maybe that dont help the feature to work correctly ..

Vidalcris, 2017-07-21:
Ok so i increased my P and now with crash_gthreshold = 500 it work perfectly and it detect crash wonderfully !
I have to say this feature is really awesome ^^

martinbudden, 2017-07-21:
Can you post some blackbox logs, both of it working well and it not working well.

Vidalcris, 2017-07-21:
I'm doing those test on my 2inch quad and there is no memory for BB unfortunately :-/
I will try with my 5inch tomorrow but i'm afraid ;)

Dont you think that the problem is the time necessary to have the gyro reflecting the command ?

martinbudden, 2017-07-21:
The crash detection works on the error rate, that is the difference between the commanded rate and the rate measured by the gyro. So increasing P should lessen the problem, as you pointed out.

Note that if you `set crash_recovery = BEEP` then the beeper will sound when the quad enters crash recovery mode, so you can tell if the mode is activated. So you could set the threshold high do  some fast flips and then reduce the threshold until the quad beeps when you do your flips. That should be a reasonably safe testing procedure - but do take care.

I don't really understand your point about time.

Vidalcris, 2017-07-21:
About time, i mean there is a delay between the command and the rate mesured by the gyro.
I mean there will always be a delay as there is always an error like you said ;)

Vidalcris, 2017-07-22:
Oh you just updated the default crash_gthreshold to 500 :) And also verify that accelerometer is on so i see that my report was usefull i'm happy XD lol
Seriously i feel better if you think that 500 is a good value for crash_gthreshold !

martinbudden, 2017-07-22:
Yes, your report was useful, thanks.

I'm always happy to make updates based on user feedback.

Vidalcris, 2017-07-22:
Just another question :
Here is a BB exemple where we can see that the error on this quad is around 700deg/s : 
https://image.ibb.co/dGKJX5/2017_07_22_11_49_19_Betaflight_Blackbox_Explorer.jpg

Can we say that for now a quad with a big error like this will not go in crash recovery because we will have to put a crash_gthreshold around 800.
Maybe a crash_delay(ms) for crash detection could work ?
As an exemple because its hard to explain for me : if the error is 800deg/s during less than crash_delay(ms) the crash recovery will not be activated.
For the BB screenshot on this exemple crash_delay = 50 allow us to reduce crash_gthreshold to 200 or even smaller :)

(Also i think some freestyler have even bigger error because they dont want a high P, the feature is not working for them for now if i'm not wrong ..)

Vidalcris, 2017-07-22:
What do you think about this : (i'm a noob so dont be rude XD)

> // if crash recovery is on and accelerometer enabled then check for a crash
            if (pidProfile->crash_recovery && sensors(SENSOR_ACC)) {
                    // Here this is only the test for crashDtermThreshold i dont really know how this part work ;)
                if (motorMixRange >= 1.0f && ABS(delta) > crashDtermThreshold) {
                    inCrashRecoveryMode = true;
                    crashDetectedAtUs = currentTimeUs;
                    inCrashRecoveryMaybe = false;
                    if (pidProfile->crash_recovery == PID_CRASH_RECOVERY_BEEP) {
                        BEEP_ON;
                    }
                    // inCrashRecoveryMode = true only if the error is longer than crashTimeLimitMaybeUs
                if (motorMixRange >= 1.0f && ABS(errorRate) > crashGyroThreshold) 
                    && (cmpTimeUs(currentTimeUs, crashDetectedMaybeAtUs) > crashTimeLimitMaybeUs) {
                    inCrashRecoveryMode = true;
                    crashDetectedAtUs = currentTimeUs;
                    inCrashRecoveryMaybe = false;
                    if (pidProfile->crash_recovery == PID_CRASH_RECOVERY_BEEP) {
                        BEEP_ON;
                    }
                    // start to measure time if error is greater than crashGyroThreshold
                if (motorMixRange >= 1.0f && ABS(errorRate) > crashGyroThreshold) {
                    inCrashRecoveryMaybe = true;
                    crashDetectedMaybeAtUs = currentTimeUs;
                    }
                    // reset inCrashRecoveryMaybe if error become smaller than crashGyroThreshold
                if (motorMixRange >= 1.0f && ABS(errorRate) < crashGyroThreshold)
                    && inCrashRecoveryMaybe = true {
                    inCrashRecoveryMaybe = false;
                    }
            }

martinbudden, 2017-07-23:
@Vidalcris , can you attach the full blackbox log for this flight?

Vidalcris, 2017-07-23:
@martinbudden Sure here it is : https://www.dropbox.com/s/za60385lhb7hn60/LX5%20Log.BFL?dl=0

This is a old log of my LX5 with F60 2450kv - ESC30A (pretty heavy build around 420g without lipo)
I sold this model some weeks ago.

Yamaford, 2017-07-25:
Raised gyro thold to 400 rates are 90RC/80Srates. flew this morning and it did fine for 2 packs until I tried yaw left roll right real hard. It didn't like that. I raised it to 500 but lowered srates a little just because they needed to slow down a tad bit anyway. I'm still tuning for 5s now so I don't have the rates exactly where I want them yet.

Long story short, 500 works. I haven't hit any trees but I did hit the ground going pretty fast. It cartwheeled then viola! I was upright again. I tried to reproduce it but going slow usually results in upside down mode with no room to self level. Maybe too much delay before it kicks in? My last crash was from a tree top that wound up bending a prop. I think it tried to self right but the power of the 5s melted the hub but I could here it trying all the way to the ground.

Vidalcris, 2017-07-25:
With the latest BF3. 2 you should be able to lower threshold to 200.
With throttle threshold =350.

_Sent from my HONOR PLK-L01 using [FastHub](https://play.google.com/store/apps/details?id=com.fastaccess.github)_

closed 2017-07-29 by Vidalcris