betaflight/betaflight #3592: Baro doesn't work on BlueJayF4

andrejpodzimek, 2017-07-22:
I soldered the optional baro sensor onto the BlueJayF4 board, [this one](https://www.aliexpress.com/item/MS5611-01BA03-SENSOR-new-and-original-free-shipping-500BP/32460242247.html?spm=a2g0s.9042311.0.0.qoa9Ia). However, it just doesn't work in Betaflight. Baro is on in Configuration, `baro_hardware` is set to `MS5611` (it didn't work with `auto` either). After an all-nighter with a multi-meter, I didn't find anything wrong as far as soldering is concerned. How does one debug this? Status doesn't show baro. Logging of altitude shows 0 all the time.

I wanted to check whether baro would work in iNavFlight (because they also have i2c magnetometers enabled for BlueJayF4 and other code differences), but iNavFlight is utterly unusable on BlueJayF4; the board just doesn't conect over USB. (Probably the MSP port is misconfigured by default.)

andrejpodzimek, 2017-07-22:
And BTW, there's no BARO among tasks.

andrejpodzimek, 2017-07-22:
I've just built and flashed the current master. The baro still doesn't work, but I noticed a few extra options in the CLI (`baro_bustype`, `baro_i2c_device`, `baro_i2c_adress`). The address was 119 by default. I also tried 118, because there's a pin that can change the least significant bit, according to the datasheet. (I think that pin is low, because GND is connected to it inside the BlueJayF4.) I tried all the 6 `baro_i2c_device` options, but nope, no sign of baro support anywhere.

Some web resources say that the address should be actually [238 (0xEE)](https://wiki.paparazziuav.org/wiki/Module/Meas_Spec_MS5611_I2C) of 239 (0xEF) But in that case I'm not sure how to configure it, because Betaflight enforces a limit of 119.

jflyper, 2017-07-22:
@andrejpodzimek If flashed 3.2, checkout if `resource list` contains any `I2Cx` lines. If there is, then set `baro_i2c_device` to `x`. `baro_i2c_address` is 118 or 119. (0xee and 0xef are 8-bit representation of 0x76/118 and 0x77/119.)

blckmn, 2017-07-22:
Also look in configurator bar for any i2c errors

andrejpodzimek, 2017-07-22:
i2c errors show 0 all the time.

In `resource` I can see the following:

    resource I2C_SCL 1 B06
    resource I2C_SDA 1 B07

However, `resource list` shows this (no `I2C_SDA` anywhere):

    B06: I2C_SCL 1
    B07: FLASH_CS

This looks surprising. Could it be a resource conflict of some sort?

jflyper, 2017-08-04:
Problem solved.

Ref: #3732

closed 2017-08-04 by jflyper