betaflight/betaflight #3599: BF 3.2 - CL_RacingF4-Buzzer makes motors stutter when armed

Zaniix, 2017-07-22:
Dhsot 1200
Armed motors then hit buzzer switch, motors stutter
Dshot Buzzer should be disabled when armed, but it appears to try and beep the motors even when armed.
I think this could be a serious issue in flight if someone accidentally hits the buzzer switch

maverikou, 2017-07-23:
Confirming this happens.

mxnew2004, 2017-07-30:
I have the dys f4 board.  I can confirm that when you activate the beeper my motors twitch while not armed.  It's as if the beacon is being activated by the buzzer.  I have not noticed the motor twitching issue while the quad is armed.

Zaniix, 2017-08-12:
on RC3, noticed motors twitching when beeper for low battery alarm is enabled. I witnessed another person having same issue. No issues if you disable the beeper for any alarms that might go off while in flight.  I do not know if this could damage a motor or ESC.

DieHertz, 2017-08-13:
@mxnew2004
That's intended behavior, you have BLHeli_S version with DSHOT commands (including beeper) support flashed onto your ESCs.
What the other guess experience is a different thing, probably a clash between PWM beeper timebase and one of the motors.