betaflight/betaflight #3605: BF3.2 My baro and mag are not detected anymore

jflyper, 2017-07-23:
There has been a drastic change in the way that baro and mag are configured and detected. This change was necessary to make baro and mag configuration more flexible. Unfortunately, the change made it hard to maintain full backward compatibility with earlier configurations.

If you experience your baro and/or mag not being detected after updating to pre-3.2 (build 1476 and later), please report here, together with following information.

- Target name.
- Baro or compass device type (e.g., BMP280, MS5611, HMC5883, AK8963, etc.)
- Whether the device is onboard or externally connected.
- If known, how the device is connected (e.g., I2C, SPI, slave on MPU)

Thank you for your contribution to make Betaflight better.

andrejpodzimek, 2017-07-23:
I’ve jist had quite a bitter experiments with 2 of the devices you mentioned, both on BlueJayF4.

I reported a baro problem in #3592. But now I tried to flash old Betafight 3.0.1 and baro still didn’t work, so it’s probably unrelated.
- Type: MS5611
- Externally connected: depends on the definition (soldered onto the board’s 8 pads meant for that purpose)
- Bus: I2C (PS pin is connected to 3.3 V inside the board, which selects I2C)

I also tried to solder [an HMC5983 module](https://www.banggood.com/3Pcs-GY-282-HMC5983-High-precision-High-sensitivity-Temperature-Compensation-p-1144414.html) (which is essentially HMC5883, just with temperature data) onto the I2C pads of my BlueJayF4, but that doesn’t work either. Then I realized that `target.h` doesn’t have the `define`s for mag sensors at all. So I looked at iNavFlight and found out that it does have mag sensors defined in BlueJayF4. But iNavFlight doesn’t work / boot on (my) BlueJayF4 and older releases are broken the same way. So I merged the BlueJayF4 `target.h` (and the list of libraries to link) from iNavFlight  into Betaflight, built the master with mag support and … nothing. No compass detected. Quite frustrating. Again, it may be unrelated to this particular problem.
- Type: HMC5983 
- Externally connected: YES (soldered onto the board’s I2C pins and 5V power; should have suitable 5V <—> 3.3V circuitry on board)
- Bus: I2C (protocol soldering pin set to I2C)

AirBreak69, 2017-07-23:
Same experience with my original REVO. I tried an old build 1169: baro works with some sporadic data errors.
I added these lines in target.h to create a custom build of build 1497.
#if defined(REVO)
#define BARO
#define USE_BARO_MS5611
#define USE_I2C
#define USE_I2C_DEVICE_1 
#define MAG
#define USE_MAG_HMC5883
#define MAG_HMC5883_ALIGN       CW90_DEG
#endif
Result: larger hex-files are generated, in CLI the baro_ and mag_ commands are now supported, but baro and mag are still not recognized.
I also tried to manually define bustype and hardware in CLI - without success.

jflyper, 2017-07-23:
@andrejpodzimek @AirBreak69 Thanks for your reports. I'm actually looking for cases that the new code doesn't detect baro and mag that were working with 3.1.7.

AirBreak69, 2017-07-24:
@jflyper I wasn't shure, what you where asking for. 3.2.0 build 1169 is prior to build 1476. So I thought it were helpfull...
But in addition I tested for you on my REVO all versions i found on my PC, just starting with defaults plus mag and baro enabled:
3.1.6: Mag and baro working fine. No sporadically disrupted data.
3.1.7: Mag and baro working fine. No sporadically disrupted data.
3.2.0 build 1121: Mag and baro working fine. No sporadically disrupted data.
3.2.0 build 1156: Mag and baro working fine. No sporadically disrupted data.
3.2.0 build 1159: Mag and baro working fine. No sporadically disrupted data.
3.2.0 build 1169, defaults: Mag and baro working fine. No sporadically disrupted data.
3.2.0 build 1169, 8K/8K/dshot600/ibus: Mag and baro working fine. rare glitches on mag.
3.2.0 build 1475: Mag and baro are not detected
3.2.0 build 1476: Mag and baro are not detected
3.2.0 build 1496: Mag and baro are not detected
3.2.0 build 1501: Mag and baro are not detected

jflyper, 2017-07-25:
@AirBreak69 Your board is not a REVO, but some AIRBOTF4 or FLIP32-F4, isn't it?

AirBreak69, 2017-07-25:
@jflyper Sorry, I do have these FCs: NAZE, SPRACINGF3EVO, AIRBOTF4, DYSF4PRO and REVO.
The REVO is an original CC3D Revolution - original as chinese warez can be ;-)

jflyper, 2017-07-25:
@AirBreak69 

Oh... it's my bad. Baro and mag support for REVO had been accidentally dropped back on Jun 16, 2017, as a part of #3229. I'm going to resurrect them. See #3624.

AirBreak69, 2017-07-26:
@jflyper Please look at #3624: You have to correct a typo in that PR in order to get baro and mag back into life on a REVO.
Tried this with a custom build using 8K/8K/dshot600/ibus: Sensor are working, but some glitches can be found in the sensors view (may be specific to my board).

jflyper, 2017-07-26:
Where is the typo?

AirBreak69, 2017-07-26:
@jflyper There's small but significant typo in line 239 of target\REVO\target.h:
Change
#define I2C2_SDA PB9
into
#define I2C1_SDA PB9

jflyper, 2017-07-26:
@AirBreak69 Fixed.

> Tried this with a custom build using 8K/8K/dshot600/ibus: Sensor are working, but some glitches can be found in the sensors view (may be specific to my board).

Can you try other settings? WIth and without turning on your TX.

AirBreak69, 2017-07-27:
Thanks so far, no probs with the current PR. Unfortunately the board currently is "copterless".
Maybe i can do some lab tests the next days...

KenImhof, 2017-08-03:
I am also having this problem with not finding the Mag on both a SP3 Deluxe fc with built-in Mag and with an M8N external Mag connected to Omnibus F4 fc.  I have verified problem exists in both 3.1.7 and 3.2RC2.

I have also confirmed that iNAV correctly finds the Mag with the exact same hardware configuration.

jflyper, 2017-08-03:
@KenImhof My apologies. Your problem is not the problem we are discussing here, as yours are not detecting them with 3.1.7 either. I suspect you are not enabling your MAG on configuration tab.

KenImhof, 2017-08-03:
I have selected Mag on the configuration tap and it is connected to UART-3 on the Omnibus F4 FC.

KenImhof, 2017-08-03:
I have selected Mag on the configuration tap as listed below and it is connected to UART-3 on the Omnibus F4 FC.

jflyper, 2017-08-03:
@KenImhof Do you still have a problem?

KenImhof, 2017-08-03:
Yes, this is still a problem.

 

Thanks,

AirBreak69, 2017-08-03:
@jflyper Looks simular to my REVO problem - you remember?
Looking at INAVs target.h of the SPRACINGF3 i find:
#define USE_I2C
#define I2C_DEVICE (I2CDEV_1) // PB6/SCL, PB7/SDA
On REVOs target.h you had to add/activate the pin definitions:
#define USE_I2C_DEVICE_1
#define I2C_DEVICE              (I2CDEV_1)
#define I2C1_SCL                PB8
#define I2C1_SDA                PB9
On SPRACINGF3s target.h the pin definitions are missing too:
#define USE_I2C
#define USE_I2C_DEVICE_1
#define I2C_DEVICE              (I2CDEV_1)

@KenImhof Can you try to add behind the I2C definition of your local SPRACINGF3s target.h:
#define I2C1_SCL                PB6
#define I2C1_SDA                PB7
and build and test a custom version?

KenImhof, 2017-08-03:
I will give this a try this evening.

 

Thanks,

KenImhof, 2017-08-03:
Adding those two lines to the SPRACINGF3 target.h did fix the problem.  The MAG is now working.

 

Thanks,

jflyper, 2017-08-04:
@KenImhof 
Since you have issues in both 3.1.7 and 3.2.0, they are not the subjects here.
Here are tips, and if you can't resolve the issues, please relocate to respective threads in RCG.

**OMNIBUSF4**
> I have selected Mag on the configuration tap as listed below and it is connected to UART-3 on the Omnibus F4 FC.
- 3.1.7 does NOT support I2C on UART3/I2C headers.
- 3.2 does support I2C on UART3/I2C headers. Enter the following commands into CLI.
```
resource i2c_scl 2 b10
resource i2c_sda 2 b11
```
Be sure NOT to enable anything for UART3.

If this doesn't work, the RCG thread is
Omnibus F4 AIO: 
https://www.rcgroups.com/forums/showthread.php?2754926-Omnibus-F4-AIO
Omnibus f4 Pro:
https://www.rcgroups.com/forums/showthread.php?2801694-Omnibus-f4-pro

**SPRACINGF3**
Your case is a pure support case. As onboard mag and baro are both working without any problem with 3.1.7 and 3.2 here locally with a Deluxe test board, I suspect your device is broken.
[image]
[image]

There's probably nothing people can do make it work, but this is the RCG thread.
https://www.rcgroups.com/forums/showthread.php?2387038-SPRacingF3-Racing-board

jflyper, 2017-08-04:
@KenImhof 
Since you have issues in both 3.1.7 and 3.2.0, they are not the subjects here.
Here are tips, and if you can't resolve the issues, please relocate to respective threads in RCG.

**OMNIBUSF4**
> I have selected Mag on the configuration tap as listed below and it is connected to UART-3 on the Omnibus F4 FC.
- 3.1.7 does NOT support I2C on UART3/I2C headers.
- 3.2 does support I2C on UART3/I2C headers. Enter the following commands into CLI.
```
resource i2c_scl 2 b10
resource i2c_sda 2 b11
```
Be sure NOT to enable anything for UART3.

If this doesn't work, the RCG thread is
Omnibus F4 AIO: 
https://www.rcgroups.com/forums/showthread.php?2754926-Omnibus-F4-AIO
Omnibus f4 Pro:
https://www.rcgroups.com/forums/showthread.php?2801694-Omnibus-f4-pro

**SPRACINGF3**
> Adding those two lines to the SPRACINGF3 target.h did fix the problem.  The MAG is now working.

I don't understand that you had to do the custom build to make it work, because both mag and baro are working without a problem here locally with both 3.1.7 and 3.2. I suspect you didn't do the full chip erase.

[image]
[image]

This is the RCG thread, just in case.
https://www.rcgroups.com/forums/showthread.php?2387038-SPRacingF3-Racing-board

KenImhof, 2017-08-04:
The following resource settings fixed the MAG not being recognized by the Omnibus F4 Pro FC:  
resource i2c_scl 2 b10
resource i2c_sda 2 b11
 
The MAG was also fixed on the SP3 Deluxe FC by adding these two lines the SPRACINGF3 target.h file:
#define I2C1_SCL PB6
#define I2C1_SDA PB7
 
Thanks,  

 

https://www.rcgroups.com/forums/showthread.php?2754926-Omnibus-F4-AIO
https://www.rcgroups.com/forums/showthread.php?2801694-Omnibus-f4-pro

 <https://user-images.githubusercontent.com/14850998/28950184-bf8b5b8c-78fc-11e7-95c2-1b7de7a8603b.jpg> 
 <https://user-images.githubusercontent.com/14850998/28950206-e4539e8e-78fc-11e7-8765-35498527e781.jpg> 

https://www.rcgroups.com/forums/showthread.php?2387038-SPRacingF3-Racing-board

jflyper, 2017-08-13:
@KenImhof 
>The following resource settings fixed the MAG not being recognized by the Omnibus F4 Pro FC:
>```
>resource i2c_scl 2 b10
>resource i2c_sda 2 b11

Actually, this is not a fix, but an intended operation that is "_Required action to enable I2C2 on OMNIBUS F4 families_".

>The MAG was also fixed on the SP3 Deluxe FC by adding these two lines the SPRACINGF3 target.h file:
>```
>#define I2C1_SCL PB6
>#define I2C1_SDA PB7

This shouldn't be necessary, as these are default pins for I2C1.

wx4cb, 2017-12-02:
omnibus F4 Nano (Flip-32) using the Lux4OSD target, the baro is a bmp280 as far as i know, link to product page on RTFQ is http://www.readytoflyquads.com/flip-32-f4-omnibus-nano

Brand new board, came already flashed with 3.2.2 looks like i can't go back because it only lists 3.2.0 as a the earliest version.

is this board just too new or ?

mikeller, 2017-12-02:
@wx4cb: See https://github.com/betaflight/betaflight/issues/1582#issuecomment-348726757 . Also, please do not cross-post your issues in multiple places. Find the issue where it fits best (or open a new issue), and post there. Having the same question in multiple places will just make it harder to have a discussion.

wx4cb, 2017-12-02:
@mikeller sorry, i'm relatively new to all this github stuff in this sense.

andrejpodzimek, 2018-06-03:
The compass on `SPRACINGF3` no longer works on Betaflight 3.3.3.

mikeller, 2018-06-03:
@andrejpodzimek: Weird, there were no changes specific to SPRACINGF3 between 3.3.2 and 3.3.3. Can you open a new issue for this please?

L3M4R, 2018-10-25:
Hi 

No luck with the HobbyWing XRotor F4 and external barometer connected to SDL/SCA (made from UART3). Baro activated in CONFIGURATION tab, Betaflight 3.5.2, TARGET OMNIBUSF4SD. No other I2C devices (GPS BN-220 on UART6).

**CLI configured by:**
`resource SERIAL_TX 3 NONE
resource SERIAL_RX 3 NONE

resource i2c_scl 2 B10
resource i2c_sda 2 B11

set baro_bustype = I2C
set baro_hardware = BMP280
`

**baro CLI output:**
`# get baro
baro_bustype = I2C
Allowed values: NONE, I2C, SPI, SLAVE
Default value: SPI

baro_spi_device = 3
Allowed range: 0 - 5

baro_i2c_device = 0
Allowed range: 0 - 5

baro_i2c_address = 0
Allowed range: 0 - 119

baro_hardware = AUTO
Allowed values: AUTO, NONE, BMP085, MS5611, BMP280, LPS, QMP6988

baro_tab_size = 21
Allowed range: 0 - 48

baro_noise_lpf = 600
Allowed range: 0 - 1000

baro_cf_vel = 985
Allowed range: 0 - 1000

baro_cf_alt = 965
Allowed range: 0 - 1000`

mikeller, 2018-10-28:
@L3M4R:

> resource i2c_sda 2 B11

> baro_i2c_device = 0

`baro_i2c_device` should probably be 2 in this case.