betaflight/betaflight #3625: SPI part of busDevice_t should be extended with clock divisor for the device

jflyper, 2017-07-25:
Devices that are sharing an SPI bus is currently setting it's own clock divisor during initialization of the device. This is an error prone approach. MAX7456 driver tries to overcome this problem by setting/restoring the clock divisor before/after each bus access, but it restores to a pre-defined divisor which is not alway correct for other devices on the bus.

One possibility is to extended the `busDevice_t.busdev_u.spi` with a clock divisor, and set the divisor in `spiBus*` functions.

Note that care must be taken to avoid this overhead for devices that requires fast path.

ledvinap, 2017-07-25:
@jflyper : It should work quite fine for spi, changing speed should be fast and may be triggered only when necessary. Bus arbitration may be a bit of problem eventually. And bud abstraction is still quite shallowm it may be easier to improve it before touching this.

With I2C this will be much more dangerous - slow devices may go out of sync when they see fast transaction (and there is no CSN to hide it)

jflyper, 2017-07-25:
@ledvinap Yeah, we can't fiddle with I2C.

jflyper, 2017-08-04:
LL for SPI has inline functions for setting clock polarity, phase and divisor. This is great...