betaflight/betaflight #3629: Crash Recovery Occurred While OFF

jcutcher01, 2017-07-26:
Flashed betaflight_3.2.0_OMNIBUSF4SD-3-46532de91.hex on my Flip32 Omnibus F4. I tried a hard flip maneuver and the quad leveled out and wouldn't allow me to pitch or roll. I verified crash recovery is off. I flashed betaflight_3.2.0_OMNIBUSF4SD-4-b13f96d29.hex and verified in CLI that I do not have it enabled. I will update this issue report when I am able to fly again.

# version
# Betaflight / OMNIBUSF4SD 3.2.0 Jul 25 2017 / 06:12:24 (b13f96d29)

# name
name Jefe 3.2

# resources

# mixer

# servo

# servo mix

# feature
feature DYNAMIC_FILTER

# beeper

# map
map TAER1234

# serial
serial 0 64 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 0 1600 2100
aux 1 4 1 1600 2100
aux 2 28 1 1600 2100

# adjrange

# rxrange

# vtx

# rxfail

# master
set baro_hardware = NONE
set min_check = 1005
set max_check = 1995
set serialrx_provider = SPEK2048
set blackbox_device = NONE
set dshot_idle_value = 550
set motor_pwm_protocol = DSHOT600
set bat_capacity = 1300
set current_meter = NONE
set battery_meter = ADC
set disarm_kill_switch = OFF
set deadband = 5
set yaw_deadband = 5
set pid_process_denom = 1
set osd_units = IMPERIAL
set osd_vbat_pos = 2401
set osd_rssi_pos = 40
set osd_tim_1_pos = 2423
set osd_tim_2_pos = 2455
set osd_flymode_pos = 2413
set osd_throttle_pos = 225
set osd_vtx_channel_pos = 377
set osd_horizon_pos = 200
set osd_current_pos = 385
set osd_mah_drawn_pos = 353
set osd_craft_name_pos = 2049
set osd_gps_speed_pos = 218
set osd_gps_lon_pos = 82
set osd_gps_lat_pos = 65
set osd_gps_sats_pos = 51
set osd_home_dir_pos = 302
set osd_home_dist_pos = 303
set osd_compass_bar_pos = 266
set osd_altitude_pos = 247
set osd_pid_roll_pos = 423
set osd_pid_pitch_pos = 455
set osd_pid_yaw_pos = 487
set osd_debug_pos = 1
set osd_power_pos = 321
set osd_pidrate_profile_pos = 345
set osd_warnings_pos = 2441
set osd_avg_cell_voltage_pos = 76
set osd_pit_ang_pos = 257
set osd_rol_ang_pos = 289
set osd_battery_usage_pos = 392
set osd_disarmed_pos = 2283
set osd_nheading_pos = 311
set osd_nvario_pos = 279
set osd_esc_tmp_pos = 82
set osd_esc_rpm_pos = 83
set vcd_video_system = 2

# profile
profile 0

set dterm_lowpass_type = PT1
set anti_gravity_gain = 4000
set p_pitch = 65
set i_pitch = 70
set d_pitch = 36
set p_roll = 60
set i_roll = 58
set d_roll = 33
set p_yaw = 90
set i_yaw = 75

# rateprofile
rateprofile 0

set rc_rate = 92
set rc_rate_yaw = 92
set rc_expo = 25
set rc_expo_yaw = 25
set thr_mid = 60
set thr_expo = 10
set roll_srate = 84
set pitch_srate = 84
set yaw_srate = 84

Vidalcris, 2017-07-26:
There is no information about crash recovery in this dump..
There is at test verifying that the option is ON before crash detection so that is not possible imo..

Can you share the result of "get crash" ?

jcutcher01, 2017-07-26:
Correct that's why I was stumped. I was flying circle patterns so I could land and check motor temps. I had pitch roll Yaw. Then I took off again and my first freestyle test was throttle punchout and hard front flip. The quad refused and self leveled and I lost all control. I landed and retrieved my quad. It was suggested that crash recovery occurred. Unfortunately I flashed 3.1.7 back so I could fly. I will retest hopefully today. I have my quad flashed and ready to fly and it has crash recovery off.

jcutcher01, 2017-07-26:
Also my dump is from DIFF.

Vidalcris, 2017-07-26:
Mhm wait for a reply from @martinbudden because i cannot understand how this could happen ...
(If this is a diff then crash_recovery is off. so yeah ...)

jcutcher01, 2017-07-26:
I also set anti gravity to 4000. In 3.1.7 it's 4.

Vidalcris, 2017-07-26:
If you still have crash parameters in 3.1.7 you should reflash with full chip erase.

jcutcher01, 2017-07-26:
I never used the feature. When I flashed 3.2.0 I set all of my parameters again from the start. I always perform a full chip erase. However, since I am downloading the HEX file manually instead of use the configurator to download, is it possible that a manual hex flash did not perform a full chip erase. I seem to recall the switch was grayed out. I will check at lunch time today (US East).

jcutcher01, 2017-07-26:
Good news! Full chip erase solved my problem. Today I flew 3.2.0-4 with PT1, both notches 0 and dynamic filtering on. Flew awesome.

closed 2017-07-26 by jcutcher01

Vidalcris, 2017-07-27:
If u had make the "get crash" as is asked for sure the parameters were corrupted..
But anyway if that work its cool.