betaflight/betaflight #3635: Omnibus F7 and DSHOT 600-1200 Spins Two Motors

TysonChritton, 2017-07-26:
Omnibus F7 hooked up to Lumenier OPTO Silk 32bit 30A escs
Using BF Configurator 3.1.1 and 3.1.2
Using BF 3.2 firmware from factory then updated to 7/26/2017 #6 version

When running DSHOT 1200 only motors 1&2 will spin when activated in configurator.
When running DSHOT 600 only motors 3&4 will spin when activated in configurator.

Dump file attached.
[dump.txt](https://github.com/betaflight/betaflight/files/1178018/dump.txt)

jflyper, 2017-07-26:
@TyTechSci 
Just tested with the current. They spin without a problem.
Did you do "defaults" and immediately test motors WITHOUT restoring settings?

TyTechSci, 2017-07-26:
Did defaults, it put it to oneshot 125, motors spun up. None of the dshot protocols work now.

jflyper, 2017-07-26:
@TyTechSci Set it to dshot, completely power down and power up again.

TyTechSci, 2017-07-26:
I could kiss you... or betaflight you... whatever. DSHOT 1200 working now! Thank you so much. Did I miss some literature somewhere to figure this out on my own?

jflyper, 2017-07-26:
No, you didn't miss anything. I don't think there is a mention of power cycling when switching to dshot (and back) is sometimes necessary.

Happy props now :)

closed 2017-07-27 by TyTechSci