betaflight/betaflight #3641: BF 3.1.7 UART not working when powering bluetooth device after powering the FC?

swilkens, 2017-07-27:
I use a bluetooth module connected to UART3 on my BFFC3 board running at 115200 baud. The +5 supply to the Bluetooth module runs through a simple switch that allows me to power the bluetooth module just when I want to to be powered. Using the bluetooth connection, I configure the firmware using an android app on my phone (EZ-GUI), which uses MSP commands.

In the past (on a naze32 rev. 5 running cleanflight), this setup would work whatever I did. I could power the bluetooth module at the same time as the FC, I coulw power the bluetooth module later. I could restart the bluetooth module without restarting the FC - it would always work.

Now, after upgrading the copter to a BFFC3 board running Betaflight 3.1.7 - I noticed that I can only get this setup working if I power the Bluetooth module at the same time as powering the FC. Additionally, removing power from the bluetooth module and powering it back on while the FC is powered also fails to setup a working connection to the app.

The module itself seems to be working fine: regardless of what I am doing, the bluetooth connection is always working fine.

So my question: are devices on the UARTs only configured during boot? Is this a change compared to classic (pre betaflight) cleanflight?

Again, this used to work perfectly on my NAZE. Now on the BFFC it only seems to work if everything is powered on at the same time.

I've also reported this to the developer of the app, since I have no way to verify if the problem is with the app or with the firmware not handling the device on the UART coming online after the FC has been powered before.

jflyper, 2017-07-27:
@swilkens I have just tested the same setup using CRIUS_BT (same as HC-06) module and OMNIBUS (F3) board flashed with 3.2, using UART1, connecting to the BT module from regular BFC configurator using serial emulation port of BT interface.

I had no problem of connecting to the FC as long as the BT module is powered. There is no constraint on when or how the BT module is powered. Only care I took was to make sure the BT device is recognized by PC before hitting "Connect" button.

Your issue is most likely not related to BF firmware.

swilkens, 2017-07-27:
@jflyper I replicated your setup (using a pc with a bluetooth dongle in stead of my phone) and I see the same problem.

1. Power FC and Bluetooth at the same time
2. Verify that I can connect on both PC (bf configurator) and phone (EZ-GUI)
3. Unpair
4. Power off bluetooth dongle on UART1, maintain power on FC
5. Power bluetooth dongle back on
6. Pair on PC (works)
7. Attempt to connect on BF Configurator (Fails, no data for 10 seconds)
8. Same story on the phone

Fully powering down the drone and the bluetooth module and restarting, powering both at the same time, fully restores connectivity.

jflyper, 2017-07-27:
@swilkens 
> I'll also hook up a scope on the rx line to see if the module is sending data and if the FC is responding.

Scoping will help a lot.

> Thanks for verifying that it's working for you (I assume you did a similar bluetooth dongle power cycle while the FC was on?)

Yeap.

swilkens, 2017-07-27:
@jflyper scope confirms the HC-05 module is completely silent on its TX after power cycling the HC-05 while the FC is connected and powered. The module HC-05 module is completely fine if I then power cycle the FC as well as the module though. Seems something is wrong with the HC-05, I'll hook it up to a programmer and see if I can figure out what's up.

Either way, thanks for taking the time to replicate my setup - it helped me out a lot.

Closing this.

closed 2017-07-27 by swilkens

swilkens, 2017-07-27:
Actually @jflyper two small verification questions:

1. Did you power the bluetooth module from a +5 from the FC or from an external regulator? You say no constraint on how the HC-06 was powered, did you try with the on board +5 as well as an external +5?

2. Do you have a way to get the bluetooth module into AT mode and check the UART settings on your bluetooth module? Mine is on 115200,1,0 (checkable via the `AT+UART?` command in AT mode).

I switched out the bluetooth module with another one and the same thing is happening, so I'm just looking for possible causes to cross off my list here.