betaflight/betaflight #3642: [SPF3evo] resource remapping issue

libwtf, 2017-07-27:
bug or user error?

while soldering a XSR receiver to a SPF3evo /me did not think all too much.

connected s.port to the TLM- (tx2) and sbus to the PPM-pin (rx2)...so i goofed it and can't use sbus and s.port-telemtry on UART2 at the same time....idiot me :-P
but instead of re-soldering i tried out resource remapping. now i'm not sure if i just use it wrong or if this is a bug..

while BF confirms the remapping, it doesn't do it.
resource shows it how i want it (tx3 remapped to the tx2 pin),
resource list shows still the original tx2 - and indeed s.port-telemetry works on tx2 not tx3.

any insights? :-)

# diff
# Betaflight / SPRACINGF3EVO 3.1.7 Apr 3 2017 / 22:28:08 (e1c4b5c)
!-> resource SERIAL_TX 2 NONE
!-> resource SERIAL_TX 3 A14
serial 1 32 115200 57600 0 115200
profile 0
rateprofile 0
rateprofile 0
# resource
resource BEEPER 1 C15
resource MOTOR 1 A00
resource MOTOR 2 A01
resource MOTOR 3 A02
resource MOTOR 4 A03
resource MOTOR 5 A06
resource MOTOR 6 A07
resource MOTOR 7 B00
resource MOTOR 8 B01
resource MOTOR 9 B10
resource MOTOR 10 B11
resource PPM 1 A15
resource LED_STRIP 1 A08
resource SERIAL_TX 1 A09
!-> resource SERIAL_TX 3 A14
resource SERIAL_RX 1 A10
resource SERIAL_RX 2 A15
resource SERIAL_RX 3 B11
# resource list
[clipped]
!-> A14: SERIAL_TX 2
[clipped]
#

jflyper, 2017-07-27:
@libwtf Need to see the result of the mapping by `resource list` before commenting.

libwtf, 2017-07-27:
ok, makes sense then. thank you.

closed 2017-07-27 by libwtf