betaflight/betaflight #3646: 3.1.7 - SPRacing F3mini Dshot not working on channel 4

RunningPenguin, 2017-07-27:
Hello,

I build a new quad with the SPracing F3mini board and Kiss 24A ESCs. I updated the ESCs to FW 1.0.2 through FTDI cable.
Then I updated the FW of the ESCs with serial passthrough - works perfect for me when the fastmode is deactivated. Thanks for these very cool feature!

After that I enabled Dshot for testing purposes and motor 4 was not spinning and the LED didn´t light up.
I swapped the wires from ESC4 and ESC2 and the failure then appeared on the other motor/ESC.
I swapped the wires with the ESC1 and ESC3 and the failure remains everytime on channel 4 of the FC. So all ESCs and motors are working on channels 1-3 and no motor/ESC is working on channel 4 in DShot mode.

In oneshot42 and oneshot125 mode all 4 ESC and motors are working so I´m thinking that the failure is in the mapping of the DShot signals to the actual pins.

I hope these bug isn´t duplicate and you could fix it fast with my description.

Thanks

jflyper, 2017-07-28:
@RunningPenguin 
As of 3.1.7, motor 4 (PWM4) of SPRACINGF3MINI is not capable of being used as Dshot output.

There may be an alternative pin you can use. Can paste output lines from CLI command "resource list"?

RunningPenguin, 2017-07-28:
Ah thanks, didn't know that. Now it makes sense that the motor4 output didn't work with version 3.1.0 and 3.2.0 also.

Output of "resource" there were no command with "list"

# resource
resource BEEPER 1 C15
resource MOTOR 1 A06
resource MOTOR 2 A07
resource MOTOR 3 B08
resource MOTOR 4 B09
resource MOTOR 5 A02
resource MOTOR 6 A03
resource MOTOR 7 A00
resource MOTOR 8 A01
resource PPM 1 B04
resource LED_STRIP 1 A08
resource SERIAL_TX 1 A09
resource SERIAL_TX 2 A14
resource SERIAL_TX 3 B10
resource SERIAL_TX 11 A01
resource SERIAL_RX 1 A10
resource SERIAL_RX 2 A15
resource SERIAL_RX 3 B11
resource SERIAL_RX 11 A00

jflyper, 2017-07-28:
Hmm... 3.1.7 should accept "resource list". Can you try again?

RunningPenguin, 2017-07-28:
Hi yeah my fault.

# resource list
IO
A00: FREE 
A01: FREE 
A02: FREE 
A03: FREE 
A04: ADC_BATT 
A05: FREE 
A06: MOTOR 1
A07: MOTOR 2
A08: FREE 
A09: FREE 
A10: SERIAL_RX 1
A11: FREE 
A12: FREE 
A13: FREE 
A14: FREE 
A15: FREE 
B00: RX_BIND 
B01: SYSTEM 
B02: FREE 
B03: LED 1
B04: FREE 
B05: USB_DETECT 
B06: I2C_SCL 1
B07: I2C_SDA 1
B08: MOTOR 3
B09: MOTOR 4
B10: FREE 
B11: FREE 
B12: SDCARD_CS 
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C13: MPU_EXTI 
C14: SDCARD_DETECT 
C15: BEEPER 
F00: FREE 
F01: FREE 
F04: FREE 

DMA:
DMA1 Channel 1: FREE
DMA1 Channel 2: FREE
DMA1 Channel 3: FREE
DMA1 Channel 4: FREE
DMA1 Channel 5: SDCARD
DMA1 Channel 6: FREE
DMA1 Channel 7: FREE
DMA2 Channel 1: ADC
DMA2 Channel 2: FREE
DMA2 Channel 3: FREE
DMA2 Channel 4: FREE
DMA2 Channel 5: FREE

jflyper, 2017-07-28:
Ah, I wanted dshot enabled, but this serves the purpose.

Okay, try this

(1) Rewire your motor 4 to PWM8.

(2) Use the following command to remap motor 4 to PWM8

```
resource motor 8 none
resource motor 4 a1
save
```

RunningPenguin, 2017-07-28:
Thank you so much. It did the trick.
For others you have to use the port declared with "7" on the top side of the PCB this is PWM8 like marked on the bottom side of the PCB.
With the above commands and Port7 DShot is working on SPracing F3mini and 4 motors.

hydra, 2017-07-28:
could you provide a config dump so that others reading this can see what you did?

RunningPenguin, 2017-07-28:
I ignored one warning that A01 is already used by Serial_TX 11.

soldered ESC/Motor4 to PWM8 (label mismatch on the PCB top side).
reset to default.
`resource motor 8 none
resource motor 4 a1
save`

it gave me a message that A01 is already used by "SERIAL_TX 11" ignored that and worked with DShot150, DShot300, DShot600.

Here is the dump.

# dump
# Betaflight / SPRACINGF3MINI 3.1.7 Apr  3 2017 / 22:28:52 (e1c4b5c)
name -
resource BEEPER 1 C15
resource MOTOR 1 A06
resource MOTOR 2 A07
resource MOTOR 3 B08
resource MOTOR 4 A01
resource MOTOR 5 A02
resource MOTOR 6 A03
resource MOTOR 7 A00
resource MOTOR 8 NONE
resource MOTOR 9 NONE
resource MOTOR 10 NONE
resource MOTOR 11 NONE
resource MOTOR 12 NONE
resource SERVO 1 NONE
resource SERVO 2 NONE
resource SERVO 3 NONE
resource SERVO 4 NONE
resource SERVO 5 NONE
resource SERVO 6 NONE
resource SERVO 7 NONE
resource SERVO 8 NONE
resource PPM 1 B04
resource PWM 1 NONE
resource PWM 2 NONE
resource PWM 3 NONE
resource PWM 4 NONE
resource PWM 5 NONE
resource PWM 6 NONE
resource PWM 7 NONE
resource PWM 8 NONE
resource LED_STRIP 1 A08
resource SERIAL_TX 1 A09
resource SERIAL_TX 2 A14
resource SERIAL_TX 3 B10
resource SERIAL_TX 4 NONE
resource SERIAL_TX 5 NONE
resource SERIAL_TX 6 NONE
resource SERIAL_TX 7 NONE
resource SERIAL_TX 8 NONE
resource SERIAL_TX 9 NONE
resource SERIAL_TX 10 NONE
resource SERIAL_TX 11 A01
resource SERIAL_TX 12 NONE
resource SERIAL_RX 1 A10
resource SERIAL_RX 2 A15
resource SERIAL_RX 3 B11
resource SERIAL_RX 4 NONE
resource SERIAL_RX 5 NONE
resource SERIAL_RX 6 NONE
resource SERIAL_RX 7 NONE
resource SERIAL_RX 8 NONE
resource SERIAL_RX 9 NONE
resource SERIAL_RX 10 NONE
resource SERIAL_RX 11 A00
resource SERIAL_RX 12 NONE
[... 403 lines cut ...]
set d_alt = 0
set p_level = 50
set i_level = 50
set d_level = 100
set p_vel = 55
set i_vel = 55
set d_vel = 75
set level_sensitivity = 55
set level_limit = 55
rateprofile 0
rateprofile 0

set rc_rate = 100
set rc_rate_yaw = 100
set rc_expo = 0
set rc_yaw_expo = 0
set thr_mid = 50
set thr_expo = 0
set roll_srate = 70
set pitch_srate = 70
set yaw_srate = 70
set tpa_rate = 10
set tpa_breakpoint = 1650

#

jflyper, 2017-07-28:
@RunningPenguin 
> Thank you so much. It did the trick.
Good to hear that. Happy props now :)

@hydra 
Timer assignments may need some review for SPRF3MINI. There are too many DMA collisions.
```
    DEF_TIM(TIM16, CH1, PA6,  TIM_USE_MOTOR, 1),  // PWM1 - PA6 // D(1,3)
    DEF_TIM(TIM17, CH1, PA7,  TIM_USE_MOTOR, 1),  // PWM2 - PA7 // D(1,1)
    DEF_TIM(TIM4,  CH3, PB8,  TIM_USE_MOTOR, 1),  // PWM3 - PB8 // D(1,5)
    DEF_TIM(TIM4,  CH4, PB9,  TIM_USE_MOTOR, 1),  // PWM4 - PB9 // ------
    DEF_TIM(TIM15, CH1, PA2,  TIM_USE_MOTOR, 1),  // PWM5 - PA2 // D(1,5)
    DEF_TIM(TIM15, CH2, PA3,  TIM_USE_MOTOR, 1),  // PWM6 - PA3 // ------
    DEF_TIM(TIM2,  CH1, PA0,  TIM_USE_MOTOR, 1),  // PWM7 - PA0 // D(1,5)
    DEF_TIM(TIM2,  CH2, PA1,  TIM_USE_MOTOR, 1),  // PWM8 - PA1 // D(1,7)
```

libwtf, 2017-08-04:
same goes for SPf3 evo (the one with SD-slot), no dshot on motor4

jflyper, 2017-08-04:
@libwtf 
Connect your motor 4 to motor 5 output (PWM5) and issue the following command.

```
resource motor 4 a6
save
```

agropom, 2017-10-17:
Dear all,
on my SPF3 Evo I installed last BF3.2.1.
I'm using RS30A lite blheli_s with oneshot125 all 4 motor spin correctly, if use dshot300 only motor 1,2,3 will spin.
I read that is necessary remap motor 4 on another "pin" but I'm having some issue I receive this message:

A06 already assigned to SERIAL RX11

Can someone help me to remap the 4th motor?

> # resource list
IO
A00: MOTOR 1
A01: MOTOR 2
A02: MOTOR 3
A03: MOTOR 4
A04: ADC_BATT
A05: FREE
A06: FREE
A07: FREE
A08: TRANSPONDER
A09: SERIAL_TX 1
A10: FREE
A11: FREE
A12: FREE
A13: FREE
A14: FREE
A15: SERIAL_RX 2
B00: FREE
B01: FREE
B02: ADC_RSSI
B03: SPI_SCK 1
B04: SPI_MISO 1
B05: SPI_MOSI 1
B06: I2C_SCL 1
B07: I2C_SDA 1
B08: LED 1
B09: MPU_CS
B10: SERIAL_TX 3
B11: FREE
B12: SDCARD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C13: MPU_EXTI
C14: SDCARD_DETECT
C15: BEEPER
F00: FREE
F01: FREE
F04: FREE

DMA:
DMA1 Channel 1: MOTOR 3
DMA1 Channel 2: TRANSPONDER
DMA1 Channel 3: FREE
DMA1 Channel 4: FREE
DMA1 Channel 5: MOTOR 1
DMA1 Channel 6: FREE
DMA1 Channel 7: MOTOR 2
DMA2 Channel 1: ADC
DMA2 Channel 2: FREE
DMA2 Channel 3: FREE
DMA2 Channel 4: FREE
DMA2 Channel 5: FREE

jflyper, 2017-10-23:
Read this thread. If you still have a issue, please redirect it to other places on the net.

closed 2017-10-23 by jflyper