betaflight/betaflight #3650: Per use based allocation of serial buffers

jflyper, 2017-07-28:
Issue status: For discussion

Some observations:
1. Current serial buffer allocation size are uniform on both rx and tx, and the size follows the largest of all possible rx or tx buffer requirements.
2. Buffer allocation is done in device level during `init`.
3. Half-duplex or simplex applications don’t need one of the rx or tx buffers.
4. Serial RX protocols with ISR callback doesn’t even need a device level rx buffer (unless RX DMA is used).
5. Some rx buffer sizes used by batch receive tasks can be calculated based on <input bandwidth> / <receiver task frequency> (or by similar formula).

ledvinap, 2017-07-28:
One possibility is to require caller of serialOpen to provide own buffers. But that will reduce memory usage only when there isn't too much modules using serial ports ...

jflyper, 2017-07-29:
> One possibility is to require caller of serialOpen to provide own buffers.

It's one way of doing it.

> But that will reduce memory usage only when there isn't too much modules using serial ports ...

But the buffers still has to be allocated dynamically within the serial port user.

8. Who allocates the buffer: (1) users of serial ports (2) `openSerial`, being passed with size request, or  (3) device level open, being passed with size request.

jflyper, 2017-07-29:
9. Current serial port users and buffer requirements (Needs amendment)

| Module | Rx | Tx | Note |
|---------|---|---|------|
| Blackbox |   |   |   |
| CLI |   |   |   |
| GPS |   |   |   |
| SmartAudio | 11  | 7  |   |
| Tramp | 16 | 16  |   |
| MSP Serial |   |   |   |
| CRSF | 0  | 36  |  ISR callback, CRSF_FRAME_SIZE_MAX |
| IBUS | 0  | 6  | ISR callback  |
| JETIEXBUS | 0 | 0 | ISR callback, open with MODE_RXTX then immediately switches to MODE_RX  |
| SBUS | 0  |  0 (telemetry?) | ISR callback  |
| Spektrum | 0  |  21 | ISR callback, SPEK_FRAME_SIZE + SRXL_FRAME_OVERHEAD   |
| SUMD | 0   | 0 (telemetry?)  | ISR callback  |
| SUMH |  0 |  0 (telemetry?) | ISR callback   |
| XBUS |  0 |  0 (telemetry?) | ISR callback   |
| FrSky (telemetry) | ---  | var  |   |
| HoTT (telemetry) | ?  | ?  |   |
| IBUS (telemetry) | ?  | ?  |   |
| LTM | ---  | var  |   |
| Mavlink |   |   |   |
| SmartPort |   |   |   |

martinbudden, 2017-07-29:
@jflyper , sorry, I think I overwrote your GPS TX value in the table.

digitalentity, 2017-07-29:
Are we going to end up having `malloc` and heap?

martinbudden, 2017-07-29:
The idea was just to use `malloc` in the initialisation. However I agree that this is potentially a lot of work, may introduce errors, and will use more FLASH. Something to investigate for 3.3.

My personal view is that we can save a lot of RAM by more careful setting of buffer sizes etc, and we should do that first. If we still find we don't have enough RAM then we can look into `malloc`.

An initial look shows that currently we are very wasteful in our buffer sizes, so quite a bit of RAM can be saved without too much effort. We should certainly initially focus on this.

jflyper, 2017-07-29:
It's too early to talk about implementation details, but I was thinking of managing serial buffer by equally divided blocks (say, 64B, by looking at the table above), and use an allocation bitmap for simplicity. Drawback of the bitmap approach is that a user has to remember how much memory it claimed, but this could be handled entirely with in `serial.c`.

closed 2020-05-09 by stale[bot]